e2nIEE/pandapower #593: pd.read_excel pull in cell value as geodata tuple

ottisredds10, 2019-12-27:
I have a excel workbook that I am attempting to pull in bus data from. I'm able to perform the create_bus function grabbing each of the columns using df.at[index, "column_name"]. A get the following UserWarning when trying to add geodata information:

"UserWarning: geodata must be given as (x, y) tupel"

The cells with the geodata information is in text format with text of (X_Lat, Y_Lon).

Is there a specific format within Excel that I need to use for the read_excel function to properly read the cell values as a tuple? Or additional code needed to convert the cell value to a tuple before trying to assign it to geodata in "create_bus"?

Thanks!

ascheidl, 2020-01-16:
maybe use something like

[float(x) for x in "(1.234, 5.678)".strip("( )").split(",")]

to convert to a tuple

ottisredds10, 2020-01-16:
Thanks, using split(",") worked! I changed my assignment to the following: geodata=df.at[idx,"coords"].split(",") 

Also, I was able to get it to work by splitting the cell in Excel into two columns, one for latitude and another for longitude. I then assigned the geodata tuple in parts using:  geodata=[df.at[idx, "x_coord"], df.at[idx,"y_coord"]]. But it looks like split accomplishes this the same way and more elegantly! 

Thanks again!

closed 2020-01-16 by ottisredds10