e2nIEE/pandapower #603: Time series generates error load flow not converged at step 0

moazzamfarid, 2020-01-15:
I have been working on this tool and this error is arriving so much. I have tried to run timeseries on the net present in the documentation of tool using the code that is also present in the documentation still this error occurs. It does not say any cause or anything just load flow did not converge. I have tried to see docs for pandapower but can't find any help.Also i tried to run powerflow and that runs fine. This is a sample net i have some other nets as well that also generate same error i want to know cause and solution of this error. I would very appreciate if anyone could help. https://nbviewer.jupyter.org/github/e2nIEE/pandapower/blob/develop/tutorials/time_series.ipynb this is link where timeseries code is present.

i couldn't upload whole notebook so here is colab link https://colab.research.google.com/drive/1PlTqRvzqvKrVFIK3Pn8b3tdocMTfcNAg

FlorianShepherd, 2020-01-16:
Does the power flow converge without the time series simulation?

Have you tried scaling the load and sgens? You have a P value time series with up to 20MW:

```
   profiles['bus1_p'] = np.random.random(n_timesteps) * 20
    profiles['load1_p'] = np.random.random(n_timesteps) * 20
```

This can be the problem. Maybe these values are "too much" for the medium voltage grid.

moazzamfarid, 2020-01-16:
Yes pp.runpp(net) runs fine. This converges without timeseries but timeseries causes this error

moazzamfarid, 2020-01-16:
And I tried 0.1 in stead of 20 and it now runs ok. Thanks a lot Florian.

closed 2020-01-16 by moazzamfarid