e2nIEE/pandapower #619: simply_plotly() throws error when trying to create artificial coordinates

jacksonzche, 2020-01-22:
Tutorial on 'Plotting Pandapower Networks using Plotly - Plots without geodata' available is broken in version 2.2.0. Simple_plotly() calls create_generic_coordinates() using 'respect_separation_points=bool'. Function is still expecting 'respect_switches=bool' as relevant argument name changes in generic_geodata.py were reverted in Dec 11 2019 commit.

FlorianShepherd, 2020-01-22:
@dlohmeier do you know the reason for this?

dlohmeier, 2020-01-22:
Sorry, there was some refactoring going wrong. Should create a hotfix... it should all be fixed in #622 . Can we create a hotfix for this?

FlorianShepherd, 2020-01-22:
Yes sure, maybe together with the other bugs that @rbolgaryn found. @rbolgaryn Can you give an estimate when you can have a look at the deepcopy problem? What about #609 and #610 is this fixed or not (the PR message looks like you've fixed it)?

closed 2020-01-28 by FlorianShepherd