e2nIEE/pandapower #620: Deepcopy doesn't copy all substructures using pp v. 2.2.0

lloewer, 2020-01-22:
e.g. net._pd2ppc_lookups['branch']

import copy
import pandapower.networks as pn
import pandapower as pp

net = pn.case9()
pp.runpp(net)
net2 = copy.deepcopy(net)

print(net._pd2ppc_lookups.keys())
print(net2._pd2ppc_lookups.keys())

results to

dict_keys(['bus', 'ext_grid', 'gen', 'aux', 'branch'])
dict_keys(['bus', 'ext_grid', 'gen'])

FlorianShepherd, 2020-01-22:
There was an issue with the deepcopy function. @lthurner changed it so that it uses pp.to_json() and pp.from_json() instead. This might lead to this issue. 

@rbolgaryn wanted to fix the deepcopy bug

lloewer, 2020-01-22:
Unfortunately, 

net2 = pp.from_json_string(pp.to_json(net))

leads to the same bug (same as net2).

lthurner, 2020-01-26:
> The workaround for now, until the issue is fixed, should be net2 = pp.from_json_string(pp.to_json(net)).

That is actually exactly what is happening internally now. The deepcopy has to be done throgh conversion to JSON, because the standard deepcopy doesn't work with JSONSerializable Objects (i.e. controllers). 

The reason some variables are omitted is because to_json ignores all internal variables that start with "_" by default (Line 496):

https://github.com/e2nIEE/pandapower/blob/07455f863b679d9b7414922b772d50d889d66620/pandapower/io_utils.py#L494-L498

This is meant to reduce file sizes, because internal matrices don't have to saved to hard drive. But for deepcopy these internal variables should be preserved. @rbolgaryn can you think of a way to pass an "include_results" flag into the to_serializable function?

dlohmeier, 2020-01-27:
I resolved this the easy way (just for the hotfix) by not using the `to_json(self)` in deepcopy, but the `to_json(dict(self))` in order to include all entries. See PR #627 (I also added csv files in the MANIFEST.in there)

dlohmeier, 2020-01-27:
Should we add a comparison (pp.nets_equal) in the deepcopy-test? I always feel like those kinds of comparisons can always raise issues later on.

closed 2020-01-28 by FlorianShepherd

rbolgaryn, 2020-02-27:
I found the solution for the underlying issue, with the pull request #676