e2nIEE/pandapower #628: pandapower 2.2.1 - fix

FlorianShepherd, 2020-01-27:
In pandapower 2.2.0 are some issues we need to fix asap. These include:

- https://github.com/e2nIEE/pandapower/issues/625 todo @rbolgaryn ?
- https://github.com/e2nIEE/pandapower/issues/620 fixed by @dlohmeier 
- https://github.com/e2nIEE/pandapower/issues/619 fixed by @dlohmeier 

possibly these:
- https://github.com/e2nIEE/pandapower/issues/610
- https://github.com/e2nIEE/pandapower/issues/609
@rbolgaryn are these fixed or not? You mentioned the PRs which seemed to fix it?

dlohmeier, 2020-01-27:
#625 is done as well

rbolgaryn, 2020-01-27:
#609 and #610 are fixed

FlorianShepherd, 2020-01-27:
great. Thanks guys, then I'll can upload the fix tomorrow :+1:

dlohmeier, 2020-01-28:
We should include #629 in here. That is also a simple fix.

jhmenke, 2020-01-28:
#631

dlohmeier, 2020-01-28:
#608 was fixed by @friederikemeier

FlorianShepherd, 2020-01-28:
> #631

should be fixed / worked around by: e8556507448037a94127383cbf81900298793614

closed 2020-01-28 by FlorianShepherd