e2nIEE/pandapower #630: oos-elements in create_generic_coordinates

dlohmeier, 2020-01-28:
I just realized that when setting parts of a network out of service (line and buses), the function `create_generic_coordinates()` raises an exception. This is the code that raises the issue:

    gnet = copy.deepcopy(net)
    gnet.bus = gnet.bus[gnet.bus.in_service == True]
    graph, meshed, roots = build_igraph_from_pp(gnet, respect_switches)

If there are oos-lines in the net connecting oos-buses, they will not be neglected in the graph creation, but the respective nodes were not added to the graph before (as the bueses were already removed). So, how should this be treated? There is the option to not exclude oos-buses or to also exclude all oos-branches (i.e. also trafos). This could also just be added as a parameter to the function...

jhmenke, 2020-04-15:
That's definitely not intended behaviour.
I think there should be a general switch that defines whether to include or exclude out of service elements.

rbolgaryn, 2020-11-10:
Hi @dlohmeier ,

the function create_nxgraph has a parameter "include_oos_elements". Does the igraph function have a similar parameter?

dlohmeier, 2020-11-10:
No, it doesn't, that should be exactly the switch that Jan-Hendrik was pointing at.

KS-HTK, 2025-08-14:
Is this still relevant?
If so it would be great if a full minimal example could be provided.

heckstrahler, 2025-11-26:
Closed due to inactivity.