e2nIEE/pandapower #638: Small bug in the pandapower.create.create_load function

gdgarcia, 2020-01-29:
Hi!

The function that create loads has a minor bug in the controllable column.
For example, creating two individual loads, the first one with the controllable parameter as default and the second one with controllable=False results in:

```python
net = pp.create_empty_network()
pp.create_bus(net, 130., index=10)
pp.create_load(net, bus=10, p_mw=150.)
pp.create_load(net, bus=10, p_mw=220., controllable=False)
pp.create_load(net, bus=10, p_mw=230.)
pp.create_load(net, bus)
print(net.load[['name', 'bus', 'p_mw', 'controllable']])
```

  name  bus   p_mw controllable
0  None   10    150.0          NaN
1  None   10    220.0        False
2  None   10    230.0        False

This is due to this portion of the code:
```python
if not isnan(controllable):
    if "controllable" not in net.load.columns:
        net.load.loc[:, "controllable"] = pd.Series()

    net.load.loc[index, "controllable"] = bool(controllable)
else:
    if "controllable" in net.load.columns:
        net.load.loc[index, "controllable"] = False
```

Instead, we should write:

```python
if not isnan(controllable):
    if "controllable" not in net.load.columns:
        net.load.loc[:, "controllable"] = False

    net.load.loc[index, "controllable"] = bool(controllable)
else:
    if "controllable" in net.load.columns:
        net.load.loc[index, "controllable"] = False
```

May I make this correction and make a pull request?

Thanks.

FlorianShepherd, 2020-01-31:
Thank you @gdgarcia for the suggestion. I think it's a good idea to set the default to False instead of NAN. 

But I'm not sure if there might be a reason why it is like this. @friederikemeier do you have any idea why it is  pd.Series?

If you change it, would be so kind to change that also for other "controllable" elements, e.g., sgen and storage? Thank you!

gdgarcia, 2020-01-31:
Hi!

> If you change it, would be so kind to change that also for other "controllable" elements, e.g., sgen and storage? Thank you!

Sure, @FlorianShepherd.
If @friederikemeier is ok with that change, I'll look for other create functions with the same issue and make the corrections.

Thanks.

friederikemeier, 2020-01-31:
Hi! This is a great suggestion and enhances consistency 👍 
Thank you for your effort!
Have a nice weekend :)

closed 2020-02-04 by lthurner