e2nIEE/pandapower #640: Diagnostic Tool for case118 produces "invalid values"

ld2874, 2020-01-30:
I loaded the 118 bus test system, and run the diagnostic tool for the system:
`n = pn.case118()
pp.runpp(n)
d = pp.diagnostic(n)`

The result is as follows:
`{'invalid_values': {'trafo': [(11, 'i0_percent', -0.04494949494949, '>=0'),
   (12, 'i0_percent', -0.16565656565657, '>=0'),
   (7, 'i0_percent', -0.64444444444444, '>=0'),
   (9, 'i0_percent', -0.81616161616162, '>=0')]}}`

Also, the field `OPF_converged` is `False`.

Pandapower version: 2.2.1

Could anyone please clarify?
Thanks

FlorianShepherd, 2020-01-31:
The diagnostic tool gives you information about the values that might be wrong. Since we converted the grids from PYPOWER / MATPOWER 118 bus case, we had to make some assumptions about some values so that the results match. Some values are not given in the PYPOWER case files, e.g., the length in km of  power line or some transformer values (only impedance values are given in the ppc). This is the reason why the diagnostic report gives you "invalid" values, since i0 is per definition positive, but has to be negative to match the ppc results here. 

OPF_converged is per default False. You have to run an OPF calculation with `pp.runopp(net)` to get OPF results and the flag will be set to `True` as well.

ld2874, 2020-01-31:
Thank you for the clarification.

FlorianShepherd, 2020-01-31:
You're welcome!

closed 2020-01-31 by FlorianShepherd