e2nIEE/pandapower #641: Scrambled order of Results reported in res_bus DataFrame of IEEE 14 bus test case

ld2874, 2020-01-30:
I run the power flow for the IEEE 14 bus test system:
`n = pn.case14()`
`pp.runpp(n)`

The `res_bus` `DataFrame` of the network looks like this:
`n.res_bus`

`vm_pu  va_degree        p_mw     q_mvar`
`0   1.060000   0.000000 -215.671492  14.824383`
`1   1.045000  -4.622589  -18.300000 -24.995932`
`10  1.059881 -13.192163    3.500000   1.800000`
`11  1.057140 -13.380924    6.100000   1.600000`
`12  1.055237 -13.360426   13.500000   5.800000`
`13       NaN        NaN    0.000000   0.000000`
`2   1.010000 -12.117572   94.200000  -4.177477`
`3   1.021034  -9.555132   47.800000  -3.900000`
`4   1.022498  -8.076028    7.600000   1.600000`
`5   1.070000 -12.671266   11.200000  -0.168466`
`6   1.065741 -11.999880    0.000000   0.000000`
`7   1.090000 -11.999880    0.000000 -15.010978`
`8   1.062786 -13.262957   29.500000  -4.860757`
`9   1.056710 -13.444795    9.000000   5.800000`

Why are the rows scrambled, and how do I get this to report results in the ascending order?

Version: 2.2.1

Thanks

gdgarcia, 2020-01-31:
Hi.

If you want the res_bus (or any other pandas.DataFrame) sorted by its index, just call the method **sort_index**, like this:

```python
# do this way if you want to preserve the original n.res_bus
res_bus_sorted = n.res_bus.sort_index()

# do this way if you want the sort to take place in the original n.res_bus
n.res_bus.sort_index(inplace=True)
```

For more about the **sort_index** method: [https://pandas.pydata.org/pandas-docs/stable/reference/api/pandas.DataFrame.sort_index.html](url)

FlorianShepherd, 2020-01-31:
Probably the network is loaded from a json file, that's why it is "scrambled". As @gdgarcia already said, you just have to sort it. It doesn't affect the power flow results.

ld2874, 2020-01-31:
Thank you for the clarification.

FlorianShepherd, 2020-01-31:
You're welcome!

closed 2020-01-31 by FlorianShepherd