e2nIEE/pandapower #644: Tests failing due to changes in pd2ppc for shortcircuit calculations

dlohmeier, 2020-01-31:
For my latest PR the tests are failing in short circuit calculation due to the following line in pd2ppc:

    buses, gs, bs = aux._sum_by_group(eg_buses_ppc, y0_grid.real, y0_grid.imag)

where y0_grid is a pandas Series.
Looking at the history confused me a little, as it turned out that this is what was there once:

    if version.parse(pd.__version__) < version.parse("0.24"):
        buses, gs, bs = aux._sum_by_group(eg_buses_ppc, y0_grid.real, y0_grid.imag)
    else:
        y0_grid_np = y0_grid.to_numpy()
        buses, gs, bs = aux._sum_by_group(eg_buses_ppc, y0_grid_np.real, y0_grid_np.imag)

And previously probably without check:

    y0_grid_np = y0_grid.to_numpy()
    buses, gs, bs = aux._sum_by_group(eg_buses_ppc, y0_grid_np.real, y0_grid_np.imag)

So, which is the option we should go for? At least any of the two versions below is better than what is implemented now, as `.real` is not an attribute of any pd.Series.

dlohmeier, 2020-01-31:
EDIT: it is in module pd2ppc_zero line 203 of function _add_ext_grid_sc_impedance_zero

FlorianShepherd, 2020-02-03:
reason: pandas v. 1.0.0 is not compatible at the moment. Quick fix: add the pandas version to setup.py

dlohmeier, 2020-02-03:
All right, I changed the pandas version in the setup.py and in the travis.yml . However, we should still check if the implementation really makes sense the way it is, as is uses values from a pandas series instead of a numpy array.

@lthurner and @derrickoswald could you check this out? I think you removed a line in your last commit that might still be required...

lthurner, 2020-02-04:
Oh seems like I missed this in the review. As commented in #592, I just wanted to use .values instead of .to_numpy to get the numpy array. So my proposed solution would be:

```
y0_grid_np = y0_grid.values
buses, gs, bs = aux._sum_by_group(eg_buses_ppc, y0_grid_np.real, y0_grid_np.imag)
```

dlohmeier, 2020-02-05:
I implemented this suggestion now and merged the PR. Now this should be working fine.

closed 2020-02-05 by dlohmeier