e2nIEE/pandapower #648: Deprecation warning due to import from pandas.util.testing

dlohmeier, 2020-02-05:
In future, it shall be importing from pandas.testing . When should we implement this? I am not very sure of which is the earliest version of pandas that is compatible with the new practice. So, for backwards compatibility we might still keep it that way...

dlohmeier, 2021-12-06:
As far as I can see, pandas.util.testing has been replaced by pandas.testing in all imports in pandapower. Can we close this issue, or would you prefer to have a version check @SteffenMeinecke ?

SteffenMeinecke, 2021-12-07:
A version check would be better. As fas as I reviewed, the change of `util.testing` to `testing` came with version 0.x -> 1.x.
https://github.com/pandas-dev/pandas/blob/0.25.x/pandas/testing.py

PR #1423 (2021-12-08): Pandas testing import
All pandas.testing imports now implemented with fallback to pandas.util.testing to close #648
closed 2021-12-08 by dlohmeier