e2nIEE/pandapower #649: Issue with pandapower simulation

MabroukAhmed, 2020-02-07:
Hi, 
When I try to rum pandapower simulation for a network composed with 9 buses using the following loads (see photo), I obtained the following error. I used the standard network implemented in the package:

net = pandapower.networks.four_loads_with_branches_out()

[image]

ascheidl, 2020-02-07:
Hi,

use pp.diagnostic(net) to get a first guess what might be wrong with your grid in case pp.runpp does not converge.

In your case it shows "overloading" - scaling all loads down by factor 0.1 solves the convergence problem. (The network is a low voltage grid)

Cheers,
alex

MabroukAhmed, 2020-02-07:
Hi Alex, 

I think that I should scale all load values between 0 and 1 because I have some load values in my database bigger than 3kv.

Thanks, 
Ahmed

SteffenMeinecke, 2020-02-25:
I don't think that there is a simple way to calculate the maximum load which allowes runpp to converge.

closed 2020-02-25 by SteffenMeinecke