e2nIEE/pandapower #660: Change generator to static generator bus

MabroukAhmed, 2020-02-19:
Hi, 

I'm actually trying to use standard network defined in pandapower library IEEE30:
net = pandapower.networks.case_ieee30()
In this network, the following buses are used to represent the grid:
- bus (30 elements)
- load (21 elements)
- gen (5 elements)
- shunt (2 elements)
- ext_grid (1 element)
- line (34 elements)
- trafo (7 elements)
- poly_cost (6 elements)
- bus_geodata (30 elements)
For this example, I would like if there is a way (using python script) to convert some generator buses to static generator ones?
I tried to set the boolean variable controllable (net.gen) to false but it does not work, since the simulation gives me always the same vm_pu (for modified generator buses) after simulations (with different load values).
Any help please?
Thanks

SteffenMeinecke, 2020-02-25:
The parameter "controllable" is used for optimal power flow (runopp). If you would like to replace generators by static generators, use the toolbox function replace_gen_by_sgen().

SteffenMeinecke, 2020-02-25:
If the answers to your issues are sufficient, please close the issues.

closed 2020-03-09 by lthurner