e2nIEE/pandapower #661: convert generator to static generator

MabroukAhmed, 2020-02-19:
Hi, 

I'm actually trying to use standard network defined in pandapower library IEEE30:
net = pandapower.networks.case_ieee30()
In this network, the following buses are used to represent the grid:
- bus (30 elements)
- load (21 elements)
- gen (5 elements)
- shunt (2 elements)
- ext_grid (1 element)
- line (34 elements)
- trafo (7 elements)
- poly_cost (6 elements)
- bus_geodata (30 elements)
For this example, I would like if there is a way (using python script) to convert some generator buses to static generator ones?
I tried to set the boolean variable controllable (net.gen) to false but it does not work, since the simulation gives me always the same vm_pu (for modified generator buses) after simulations (with different load values).
Any help please?
Thanks

Amonara, 2020-02-19:
Hi,
well you may use straight forward method. Simply make a loop to take variables from gens and transfer them to sgens. Then turn gens out of service.

`buses = nw.gen.bus.values

for i in buses:
    pp.create_sgen(net, bus=net.gen.bus.loc[i], p_mw=net.gen.p_mw.loc[i])
    nw.gen.in_service.loc[i] = False`

MabroukAhmed, 2020-02-19:
Hi, 

Thank you for the response. However, we don't need to mention the q_mv when we create sgen?

Amonara, 2020-02-24:
Hi, sorry for late respond. Not necessary. Look at the PP documentation. Q_mvar will be set on 0.

closed 2020-03-09 by lthurner