e2nIEE/pandapower #666: Allow users to choose SuperLU or UMFPACK

mzy2240, 2020-02-23:
In the pandapower paper, it mentioned the pandapower was using the SuperLU as the sparse linear solver cause at that time the scipy was using that as default, however scipy has been using UMFPACK as its default solver for about two years after that. In order to compare with some stats in the paper, can you add an argument to support switching the sparse linear solver? I know scipy supports that, so it should not be hard to implement. Thanks!

FlorianShepherd, 2020-02-25:
I added the options `use_umfpack` and `permc_spec` as defined here:
https://docs.scipy.org/doc/scipy/reference/generated/scipy.sparse.linalg.spsolve.html

I pushed it to my branch and will merge to develop: 
https://github.com/FlorianShepherd/pandapower

you can call it with: `pp.runpp(net, use_umfpack=False)`
to deactive UMFPACK.

FlorianShepherd, 2020-02-25:
merged it : 908e8bf55d844b044a3bd286d49ac627e3f5cff6

If you plan to evaluate some timings I'd be interested of your results, @mzy2240  :) I measured the timings for the paper back then. I also tried to further reduce the time with the `permc_spec` function, but that never seemed to work. Matlab is quite cool here, it does a pre-analysis of it by default.

closed 2020-02-25 by FlorianShepherd

mzy2240, 2020-02-25:
@FlorianShepherd Just tested in Windows platform and no clear difference. Installed SuiteSparse using conda and installed scikit-umfpack using pip. Tested in 150, 500, 1000, 2000 and 5000 bus cases, and also with and without numba, the difference between using UMFPACK and no using UMFPACK is neglectable. Perhaps the J matrix is not big enough to make a difference? We also have 75000 bus case but that case might take a year to convert lol. Another possibility is that scipy cannot link to SuiteSparse in Windows thus it is actually running SuperLU all the time. If you have another OS, do you mind to do some tests? Thanks!