e2nIEE/pandapower #669: Correlation between voltage magnitudes

MabroukAhmed, 2020-02-25:
Hi,
I'm actually trying to make simulation (using real load data) using standard case33bw.  Once the data is generated, and when I try to plot the correlation matrix between magnitude bus voltages, I obtain the following result:
[image]

This result is pretty compatible with the network topology (buses network connections), since neighbours buses, as can be seen are characterised by very high correlations (yellow colour).

However, When I try to add a line between Bus 1 and Bus 16 in the network, and after generating new simulation data using the updated network, the correlation between bus 1 voltage magnitude and bus 16 magnitude seems to be very low instead of the line between them in the network. In addition, bus 1 magnitude, as can be seen, seems to be more correlated with bus 22
[image]

Here is the command that I used to add line in the network:
line2 = pp.create_line_from_parameters(net, from_bus = 1, to_bus = 16, df = 31, c_nf_per_km = 0, max_loading_percent=100, g_us_per_km = 0, in_service = True, length_km=3,  max_i_ka = 99999.000, parallel = 1, r_ohm_per_km = 0.5, type ='ol', x_ohm_per_km=0.23)

Do you have any explication about these results?
By the way, do you know how to get the ID of a bus in the network?
Thank you

jhmenke, 2020-02-25:
In your plots, do you show the pandapower indices of the buses or the positions inside the pandas dataframes?

It seems to work for me:

```
>>> net = nw.case33bw()
>>> net.bus
    in_service  max_vm_pu  min_vm_pu name type  vn_kv zone
0         True        1.0        1.0    0    b  12.66    1
1         True        1.1        0.9    1    b  12.66    1
10        True        1.1        0.9   10    b  12.66    1
11        True        1.1        0.9   11    b  12.66    1
12        True        1.1        0.9   12    b  12.66    1
13        True        1.1        0.9   13    b  12.66    1
14        True        1.1        0.9   14    b  12.66    1
15        True        1.1        0.9   15    b  12.66    1
16        True        1.1        0.9   16    b  12.66    1
17        True        1.1        0.9   17    b  12.66    1
18        True        1.1        0.9   18    b  12.66    1
19        True        1.1        0.9   19    b  12.66    1
2         True        1.1        0.9    2    b  12.66    1
20        True        1.1        0.9   20    b  12.66    1
21        True        1.1        0.9   21    b  12.66    1
22        True        1.1        0.9   22    b  12.66    1
23        True        1.1        0.9   23    b  12.66    1
24        True        1.1        0.9   24    b  12.66    1
25        True        1.1        0.9   25    b  12.66    1
26        True        1.1        0.9   26    b  12.66    1
27        True        1.1        0.9   27    b  12.66    1
28        True        1.1        0.9   28    b  12.66    1
29        True        1.1        0.9   29    b  12.66    1
3         True        1.1        0.9    3    b  12.66    1
30        True        1.1        0.9   30    b  12.66    1
31        True        1.1        0.9   31    b  12.66    1
32        True        1.1        0.9   32    b  12.66    1
4         True        1.1        0.9    4    b  12.66    1
5         True        1.1        0.9    5    b  12.66    1
6         True        1.1        0.9    6    b  12.66    1
7         True        1.1        0.9    7    b  12.66    1
8         True        1.1        0.9    8    b  12.66    1
9         True        1.1        0.9    9    b  12.66    1
>>> pp.runpp(net)
>>> net.res_bus.loc[[1, 16]]
       vm_pu  va_degree  p_mw  q_mvar
1   0.997032   0.014481  0.10    0.06
16  0.913698  -0.485473  0.06    0.02
>>> pp.create_line_from_parameters(net, from_bus = 1, to_bus = 16, df = 31, c_nf_per_km = 0, max_loading_percent=100, g_us_per_km = 0, in_service = True, length_km=3, max_i_ka = 99999.000, parallel = 1, r_ohm_per_km = 0.5, type ='ol', x_ohm_per_km=0.23)
>>> pp.runpp(net)
>>> net.res_bus.loc[[1, 16]]
       vm_pu  va_degree  p_mw  q_mvar
1   0.997085   0.014141  0.10    0.06
16  0.988275   0.085842  0.06    0.02
```

As you can see, the voltage at bus 16 is increased considerably by the new line.

MabroukAhmed, 2020-02-25:
Hi, 
Yes I have seen the same thing. however, I dont understand how the voltage of a bus can be more correlated to a non connected bus than its neighbour in the network. Here is an example, where I added a line between bus 23 and 32. After simulation, results show that bus 32 is more correlated to bus 22 than bus 23.
 
[image]

In the photo I show indices of the bus (not the index in pandapower). I double checked this point to be sure that I plot the right indice of each bus.

jhmenke, 2020-02-26:
can you post the code for the correlation plot?

MabroukAhmed, 2020-02-26:
Here is my code (note that res_sim contains the whole simulation for all buses):

names = [str(i) for i in range(1,33)]
col1 = ["bus_" + str(i) + '_vm_pu' for i in range(1,33)]
correlations = res_sim[col1]
correlations = correlations.corr()
#plot correlation matrix
fig = plt.figure(figsize=(20, 7))
ax = fig.add_subplot(111)
cax = ax.matshow(correlations, vmin=0.2, vmax=1)
fig.colorbar(cax)
ticks = np.arange(0,32,1)
ax.set_xticks(ticks)
ax.set_yticks(ticks)
ax.set_xticklabels(names)
ax.set_yticklabels(names)
plt.show()

You can just take a simple example without adding a new lines as I done before. Let's just consider the correlation between buses 17 and 32 (both buses are connected in the original network). As can be seen in the current plot, the correlation between them is low comparing to other such as the correlation between buses 32 and 22 (which are not connected by a line in the network).

MabroukAhmed, 2020-02-26:
Here is my code for the simulation part:

def simulation_load(net, data):
    
    pf_i = np.random.uniform(0.15, 0.95,data.shape[0])
    col1 = ["bus_" + str(i) + '_vm_pu' for i in net.bus.index]
    col2 = ["bus_" + str(i) + '_va_degree' for i in net.bus.index]
    # arc cosine(pf_i)
    # Tang of arc cosine
    # multiply with real power
    
    # create empty database
    df = pd.DataFrame(columns=countList(col1, col2))
    for i in range(0, data.shape[0]):
        array_vol = []
        for j in range(0, net.load.shape[0]):
            net.load['p_mw'].iloc[j] = data.iloc[:,j].iloc[i]
            net.load['q_mvar'].iloc[j]=( data.iloc[:,j].iloc[i]) * (math.sqrt(1 - (pf_i[i]**2))) 
        # run simulations
        pp.runpp(net)  
        # get buses results
        bus_res = net.res_bus
        for k in range(0, bus_res.shape[0]):
            array_vol.append(bus_res['vm_pu'].iloc[k])
            array_vol.append(bus_res['va_degree'].iloc[k])
        # build data frame
        df.loc[i] = array_vol    
    return df
# power_cons variables contains time series of load profiles.
res_sim = simulation_load(net, power_cons_s)

jhmenke, 2020-02-26:
Your indexing code for the results and elements in general looks correct. 

I think an analysis of the results of the correlation of power flow results is not an issue that is specific to the pandapower software. However, i'll leave it open if anyone wants to discuss further.

closed 2020-03-09 by lthurner