e2nIEE/pandapower #675: Shortcircuit: incorrect handling of generators in branch currents

chefPony, 2020-02-26:
The following simple example reproduce the behaviour

```
# Example
# Bus0    Bus1    Bus2     Bus3        sGen0
#  _  trafo  _  line0 _  line1  _
# |_|-------|_|------|_|-------|_|--------G
#
simple_net = pp.create_empty_network(f_hz=50., sn_mva=1.)
pp.create_bus(simple_net, bus=0, vn_kv=15.)
pp.create_bus(simple_net, bus=1, vn_kv=0.4)
pp.create_bus(simple_net, bus=2, vn_kv=0.4)
pp.create_bus(simple_net, bus=3, vn_kv=0.4)

pp.create_transformer_from_parameters(simple_net, hv_bus=0, lv_bus=1, vn_hv_kv=15., vn_lv_kv=0.4, 
                                      pfe_kw=0.5, vk_percent=4.0, vkr_percent=1., sn_mva=0.4, i0_percent=1.)

pp.create_line_from_parameters(simple_net, from_bus=1, to_bus=2, length_km=0.01, r_ohm_per_km=1., x_ohm_per_km=1., 
                               max_i_ka=1, c_nf_per_km=0. )
pp.create_line_from_parameters(simple_net, from_bus=2, to_bus=3, length_km=0.01, r_ohm_per_km=1., x_ohm_per_km=1., 
                               max_i_ka=1, c_nf_per_km=0. )

pp.create_sgen(simple_net, bus=3, sn_mva=0.1, k=100., p_mw=0., q_mvar=0.)
pp.create_ext_grid(simple_net, bus=0, vm_pu=1.1, s_sc_max_mva=1e30, rx_max=0.)
```

If we compute shortcircuit current **without generation**:

```simple_net.sgen.in_service = False
shortcircuit.calc_sc(simple_net, case='max', fault='3ph', branch_results=True)
simple_net.res_line_sc['ikss_ka']
print("Line")
print(simple_net.res_line_sc['ikss_ka'])
#Line
# 0    8.636747
# 1    5.885113
print("Bus")
print(simple_net.res_bus_sc['ikss_ka'])
# Bus
# 0    3.849002e+28
# 1    1.554649e+01
# 2    8.636747e+00
# 3    5.885113e+00
```

If we compute shortcircuit current **with generation**:
```simple_net.sgen.in_service = False
shortcircuit.calc_sc(simple_net, case='max', fault='3ph', branch_results=True)
simple_net.res_line_sc['ikss_ka']
print("Line")
print(simple_net.res_line_sc['ikss_ka'])
#Line
# 0    8.636747 the same as before, even if now the generator is active
# 1    5.885113 the same as before, even if now the generator is active
print("Bus")
print(simple_net.res_bus_sc['ikss_ka'])
# Bus
# 0    3.849002e+28
# 1    2.998025e+01  correctly accounts for generator contribution
# 2    2.307050e+01  correctly accounts for generator contribution
# 3    2.031887e+01  correctly accounts for generator contribution
```

Similar problem with `shortcircuit.calc_single_sc`:
```simple_net.sgen.in_service = True
shortcircuit.calc_single_sc(simple_net, bus=2, case='max', fault='3ph')
print("Line")
print(simple_net.res_line_sc['i_ka'])
# Line
# 0    8.636747e+00 same as if without generator
# 1    2.563950e-15  this should be higher since the sc current should flow from G to bus 2 where the fault happened
```

lthurner, 2020-03-02:
Thanks for reporting this, that does look like a bug. The branch results are not fully tested yet, that is why there is also a warning printed that branch results are in beta mode. I will try to look into it, but can't promise I will get around to it soon. If you want to take a shot at it, we always welcome contributions!

chefPony, 2020-03-11:
Hi Leon, 
thank you for your answer. I've given a look to the source code and in my opinion the bug lies in file `pandapower/shortcircuit/currents.py` inside the function `_calc_branch_currents`. 
First of all in line 89 ` Yt = ppc["internal"]["Yf"]` should be `Yt = ppc["internal"]["Yt"]` .
Second, in lines 99, 100:
```
ikss1_all_f[abs(ikss1_all_f) < 1e-10] = np.nan
ikss1_all_t[abs(ikss1_all_t) < 1e-10] = np.nan
```

This seems strange. Why put `np.nan` instead of `0` ? Putting `np.nan` raise problem afterwards since when you sum ikss1 and ikss2 together adding some value to `np.nan` returns `np.nan`.
I haven't tested it throughly but in the above example, it seems to work.
What do you think?

chefPony, 2020-03-31:
I' ve been looking on this, as a first step I am writing some simple tests with the following network scheme:
```
# 
# Bus0      Bus1     Bus2          
#   _  line0  _  line1 _ 
#  |_|-------|_|------|_|
#   |        |
#  EG        G
```
where `EG` is the `ext_grid` and `G ` stands in turn as a static generator, asynchronous motor and synchronous generator. These tests should pinpoint the bug and validate the fix above. 
However, I have doubts on how a potential fix for this bug should behave in a minimum short circuit calculation. Consider the following (taken from the current test suite):
```
def three_bus_example():
    net = pp.create_empty_network()
    b1 = pp.create_bus(net, 110)
    b2 = pp.create_bus(net, 110)
    b3 = pp.create_bus(net, 110)

    pp.create_ext_grid(net, b1, s_sc_max_mva=100., s_sc_min_mva=80., rx_min=0.4, rx_max=0.4)
    pp.create_line(net, b1, b2, std_type="305-AL1/39-ST1A 110.0" , length_km=20.)
    pp.create_line(net, b2, b3, std_type="N2XS(FL)2Y 1x185 RM/35 64/110 kV" , length_km=15.)
    net.line["endtemp_degree"] = 80

    pp.create_sgen(net, b2, sn_mva=2, p_mw=0, k=1.2)
    return net
```
In this simple network the resulting short circuit `line x bus` current matrix is the following:
```
bus3= three_bus_big_sgen_example()
shortcircuit.calc_sc(bus3, branch_results=True, case='min')
# resulting ikss_ka matrix: 
#                    bus0           bus1         bus2
# line_0       [0.01259673,  0.3989686 ,   0.39170662]
# line_1       [0.          ,  0.        ,  0.40431286]
```
Taking the minimum by row in the matrix above produces the technically correct but in practice useless result of `[0.01259673, 0.]`. Which is not what the user probably wants. Instead of taking the minimum I propose the following alternatives:

1. Take the minimum of all the ikss_ka line matrix entries above some threshold which could be specified by the user or taken as the line nominal current limit
2. Return directly the ikss_ka line matrix (this would apply also when `case='max'`) and let the user process it as he wants.

I think the first option would be easier to implement, however the second one is probably more general and flexible. @lthurner  let me know if you have other ideas or any suggestion.
Thanks

lthurner, 2020-08-13:
Thanks for fixing this!

closed 2020-08-13 by lthurner