e2nIEE/pandapower #682: from_mpc seems not working

LIESEACN, 2020-03-04:
When I use the pp.converter.from_mpc to convert a matpower format file, i got following alert, which is '_Matfile does not contain a valid mpc structure._' and '_KeyError : 'bus'_ '
[case24_ieee_rts.zip](https://github.com/e2nIEE/pandapower/files/4284780/case24_ieee_rts.zip)

lthurner, 2020-03-06:
@SteffenMeinecke  can you take a look at this?

SteffenMeinecke, 2020-03-09:
Regarding replace_sgen_by_gen():
This is a json converter bug.
To create a gen instead of an sgen, the `vm_pu` information is needed. However the json converter currently saves nan values to the result tables which is also done fpr the case24_ieee_rts() network.
Till this is fixed you can of course use the replace_sgen_by_gen() functionality if you fill the result tables by running a power flow before, as here:

```
import pandapower as pp
import pandapower.networks as pn

net = pn.case24_ieee_rts()
pp.runpp(net)  # this fixes the bug
pp.replace_sgen_by_gen(net)
```

SteffenMeinecke, 2020-03-09:
It seems to be no json converter bug anymore but since commit 9a1fe484a0ca2b0259bc8b53a64b405ab485dbd1 the pandapower.network.caseXXX have nan result values (v1.6.1 was the last version with result values).

SteffenMeinecke, 2020-03-09:
Regarding the from_mpc():
Within the `case24_ieee_rts.mat` file the matpower struct is named `case24_ieee_rts`, not `mpc`. So you need to give that name to the `from_mpc()` function as here:

```
import pandapower.converter as pc
import os
folder_path = r"C:\Users\user\folder"
net = pc.from_mpc(os.path.join(folder_path , "case24_ieee_rts.mat"), casename_mpc_file="case24_ieee_rts")
```

LIESEACN, 2020-03-09:
Thank you, Steffen. Your help is very useful. Thanks a lot.
But I have found another problem which is about replace_ext_grid_by_gen, the model is the case24_ieee_rts, too. After runpp(net), I replace the sgen successfully, but when i replace the ext_grid by gen, and runopp(net), it will notice that the Net doesn't include slack bus. And I found that it is because the bus where ext_grid locates doesn't convert to slack bus automatically. After type this
`net.gen.loc['ext_grid_bus','slack'] = True` .
The runopp(net) works.

SteffenMeinecke, 2020-03-10:
Yes, that is the way you can do it.
In pandapower buses with ext_grids are per default slack nodes, buses with gens per default pv nodes and buses with sgens per default pq nodes. You can make generator nodes to slack nodes via changing the value of the net.gen.slack column to True. To allow the opf to use the sgen reactive power as variable, you can set them to controllable, which is already given in case24_ieee_rts.

PR #689 (2020-03-10): resave network jsons with empty result tables and improve replace_sgen_by_gen
fixes #682
closed 2020-03-10 by SteffenMeinecke