e2nIEE/pandapower #683: runopp results cannot be recreated with runpp while using const_i_percent

jwiemer112, 2020-03-04:
If you calculate a runopp(net) and transfer the solution to the network: 

```
net.sgen.p_mw = net.res_sgen.p_mw / net.sgen.scaling
net.sgen.q_mvar = net.res_sgen.q_mvar / net.sgen.scaling

net.load.p_mw = net.res_load.p_mw / net.load.scaling
net.load.q_mvar = net.res_load.q_mvar / net.load.scaling
```

the results cannot be reacreated with the runpp(net) if youre using **const_i_percent=100** in the loads.

Following test shows off the behvaiour:

https://github.com/jwiemer112/pandapower/blob/develop/pandapower/test/opf/test_opf_pf_comparison.py

lthurner, 2020-03-06:
Yes as far as I know the OPF does not support constant current loads. I am not sure if it can be easily fixed inside the OPF, if not we could at least add a warning in the OPF that const_i_percent will be ignored...

SteffenMeinecke, 2021-11-10:
As @lthurner stated, @jwiemer112 could you please check if this issue can be easily fixed or, if not, add a warning?

SteffenMeinecke, 2021-11-16:
@jwiemer112 in the ppc structure, there is no possibility to consider `const_i_percent`. `const_z_percent` loads are similar to shunts so such could be represented in `ppc["bus"]` columns `Gs`, `Bs` (4, 5).
But according to my code review, `const_z_percent` is never integrated to the ppc/ppci structure (neither `runopp()` nor `runpp()`) but `const_i_percent`. `const_z_percent` are set to new columns 13, 14 -> `build_bus.py` `_calc_pq_elements_and_add_on_ppc()` line 390 and following.
Therefore, this issue is also relevant to the converter functions `to_ppc()` and `convert_pp_to_pm()`, since these, similar to `runopp()` rely on the usual ppc structure.

commit bfa0b10e (2021-11-16): Merge pull request #1393 from SteffenMeinecke/warn_voltage_depend_loads

fix #683 by warn missing voltage_depend_loads implementation
commit db6467dc (2021-11-16): fix #683 by warn missing voltage_depend_loads implementation
closed 2021-11-16 by SteffenMeinecke