e2nIEE/pandapower #694: qps_pypower Failing during tests

fgfgama, 2020-03-11:
Hi! I'm running

```
import pandapower.test
pandapower.test.run_all_tests()
```

As suggested by the Getting Started guide.

When it reaches pandapower/pypower/qps_pypower.py it fails.

It attempts to run the code with alg=600 (MOSEK), and because it's not one of 0, 200 or 250, it goes into the branch (line 171)

```
    else:
        sys.stderr.write('qps_pypower: %d is not a valid algorithm code\n', alg)
```

This gives the first error: "write() takes one argument, two are given". I fixed this by setting
`sys.stderr.write('qps_pypower: %d is not a valid algorithm code\n' % alg)` instead.

But when I run it next, it says output is referenced before assignment (because output is defined only in the other branch of the if clause). And this I'm not sure how to fix (if I move the output and the return into the if clause, then it tells me that the output fo that function can't be unpacked).

Any ideas on how to fix this? I'm just getting started with Pandapower, and that's why I wanted to run the tests as suggested in the documentation.

Thanks!

lthurner, 2020-03-11:
Thanks for submitting this! Can you say in which test the error originates?

I am guessing this has something to do with the installed solvers, since the tests are passing on the Travis server. Of course the tests should pass on all systems, so we should look into this.

fgfgama, 2020-03-11:
Hi lthurner,

This is the test that failed:

```
.conda/envs/basic3/lib/python3.7/site-packages/pandapower/test/opf/test_basic.py .........qps_pypower: 600 is not a valid algorithm code
F
```

Of course the "600 is not a valid algorithm code" appears because I fixed the write() function, otherwise I wouldn't be getting there either.

Thanks for your help! I just want to know that it's working before I start using it fully :)

lthurner, 2020-03-12:
I think the problem is that pypower finds that you have installed MOSEK, and therefore tries to use it. However, pypowers binding to MOSEK does not work properly, so it fails. On systems where MOSEK is not installed, it defaults to a different solver, which works. If I remember correctly we had the same issue in the AC OPF, but it seems we never fixed this in the DC OPF.

Cany ou please comment out these two lines:
https://github.com/e2nIEE/pandapower/blob/a3ef93405c97fe2df4fc4278292b9e247f0f9791/pandapower/pypower/dcopf_solver.py#L84-L85
And report back if this solves the issue? I suspect it will, but I can't reproduce it because I don't have MOSEK...

fgfgama, 2020-03-12:
Yes, thanks!

That fixed that test. So this means that Pandapower actually doesn't support Mosek? (I understand that this part was obtained from the pypower library, which I think has been deprecated, so it would make sense if Mosek is not supported by Pandapower; I just want to be sure that there's no way to use MOSEK).

I also got another error in another test. I should create a new issue ticket, right?

Thank you very much for your help!

lthurner, 2020-03-16:
> That fixed that test. So this means that Pandapower actually doesn't support Mosek? (I understand that this part was obtained from the pypower library, which I think has been deprecated, so it would [...]

No, as far as I know PYPOWER never had a working MOSEK interface, and this was never fixed in pandapower. If you are interested in getting that to work I would be glad to help pointing you in the right direction.

> I also got another error in another test. I should create a new issue ticket, right?

Yes please open another one.

closed 2020-03-16 by lthurner