e2nIEE/pandapower #699: New OPF required?

thomaswolgast, 2020-03-15:
I would like to start a discussion about the pandapower Optimal Power Flow (OPF).
In my work, I found that the utilized pypower-OPF is - although very fast and simple to apply - rather limited in its capabilities (see also issues #678 and #683).

Some limitations are that integers cannot be optimized (e.g. optimal tap-changing), that some objective functions are not possible (e.g. min sum((u-1)^2)), and that some constraints are not possible easily (e.g. S_max). And I am not even talking about more sophisticated OPFs like multi-stage optimization, security constrained OPF, or probabilistic OPF.

What is your opinion about these limitations? Do you think pandapower needs a more advanced OPF that allows to solve a greater variety of problems? I am thinking about creating and contributing such an algorithm.

How do you deal with these limitations at the moment? Do you implement your own OPF, using standard solvers or meta-heuristics? Or do you simply use other tools for OPF calculation, like PowerModels.jl? Is a further development of the pandapower-OPF maybe even pointless?
(Or are the limitations maybe not as big as I made them?)

I look forward to hearing your opinions about these thoughts, especially from the core team. Thank you!

lthurner, 2020-03-17:
You are definetly right that the pandapower OPF is not as well developed as other capabilities of pandapower. We have discussed improving the OPF many times, but until now unfortunately we did not have the time and ressources to implement a better OPF. The practical solution was to provide an interface to PowerModels, which already provides a great OPF. Of course the interface to another language is not easy to handle, and customization of the optimization problem is also not easily possible. A implementation that is closer to pandapower would definetly be preferable, but of course takes more resources than providing an interface.

> What is your opinion about these limitations? Do you think pandapower needs a more advanced OPF that allows to solve a greater variety of problems? I am thinking about creating and contributing such an algorithm.

That would definetly be very welcome. If there is a push to implement an improved OPF for pandapower we might also be able to contribute, I just don't think we have the right projects right now to drive the development completely on our own. Do you have a specific use case that you can describe that you are looking at right now?

thomaswolgast, 2020-03-18:
I think that an interface to more advanced OPF tools like PowerModels was a good idea. Effort is smaller and the resulting capabilities are better compared to implementing a new pandapower-OPF from scratch, which would have limited capabilities again.
However, regarding PowerModels, I see two main weaknesses: Julia programming language is required, and customized OPFs (defined within pandapower) are difficult. 
Approach: To solve these problems, maybe an interface to an advanced python optimization tool like [Pyomo](http://www.pyomo.org/about) would be useful. That interface would solve the two-language problem. If it is well done, it should also allow for customization of the OPF problem, because pyomo is very flexible. Alternatively, the PowerModels interface could be adapted to allow for better customization within pandapower. 

> Do you have a specific use case that you can describe that you are looking at right now?

I personally need an OPF that is able to solve all kinds of problems, because I plan to develop a methodology for real-time capable OPF in my thesis, which is applicable to arbitrary OPF algorithms. Therefore, in the long-term, I will probably need even sophisticated variants like security-constrained OPF. For that reason, customizability of the OPF is the main required feature for me personally. 

PS: Some time ago, I implemented a simple genetic algorithm for OPF that is compatible with pandapower. It allows for arbitray objective functions and constraints (but is slow as hell). In anybody interested in that code? In general, meta-heuristics provide perfect customizability.

lthurner, 2020-03-24:
>  Approach: To solve these problems, maybe an interface to an advanced python optimization tool like Pyomo would be useful. That interface would solve the two-language problem. If it is well done, it should also allow for customization of the OPF problem, because pyomo is very flexible.

Are you aware of EGRET: https://github.com/grid-parity-exchange/Egret? This provides OPF problem formulation in pyomo and could be used as a starting point for such an OPF.

> PS: Some time ago, I implemented a simple genetic algorithm for OPF that is compatible with pandapower. It allows for arbitray objective functions and constraints (but is slow as hell). In anybody interested in that code? In general, meta-heuristics provide perfect customizability and could be a solution as well.

I think this would be an application built around pandapower rather than an internal part of pandapower. You can set up a seperate repo for this code, it might certainly be useful to others!

thomaswolgast, 2020-03-29:
> Are you aware of EGRET: https://github.com/grid-parity-exchange/Egret? This provides OPF problem formulation in pyomo and could be used as a starting point for such an OPF.

I wasn't aware of EGRET. I will surely look into it. Thank you very much!

> I think this would be an application built around pandapower rather than an internal part of pandapower. You can set up a seperate repo for this code, it might certainly be useful to others!

Yes, I agree, which is why I did not propose that solution for pandapower. However, I made the code public on my [Account](https://github.com/thomaswolgast/gaopf). Feel free to look into it and play with it, or even contribute, if it is helpful for you. 

I created an examples-file that demonstrates some of the features, for example to optimize the voltage profile instead of costs.

lthurner, 2020-03-30:
> I wasn't aware of EGRET. I will surely look into it. Thank you very much!

If you look into the contributors you will see that some of the contributors are also pyomo developers, so we can be sure that these guys know what they are doing with regards to pyomo :)

>I created an examples-file that demonstrates some of the features, for example to optimize the voltage profile instead of costs.

Cool, thanks a lot!

friederikemeier, 2020-12-02:
I think we can close this due to inactivity. @thomaswolgast  We would still appreciate your contributions on enhancing the pandapower interfaces to other solvers :) Also, if you work with pandapower and other solvers, we would be glad if you shared your experience 😄

closed 2020-12-02 by friederikemeier