e2nIEE/pandapower #701: test_1ph Failing during test

fgfgama, 2020-03-16:
Hi! I think I have an error during the tests related to this error:
https://github.com/e2nIEE/pandapower/issues/644

However, I don't understand how to fix that error (which lines of code should I edit to do what) and since it is closed I thought that it was fixed already.

In any case, I'm running the tests as follows

```
import pandapower
import pandapower.networks
import pandapower.topology
import pandapower.plotting
import pandapower.converter
import pandapower.estimation
import pandapower.test
pandapower.test.run_all_tests()
```

The failed test is
`lib/python3.7/site-packages/pandapower/test/shortcircuit/test_1ph.py F` with the following error
`lib/python3.7/site-packages/pandapower/test/shortcircuit/test_1ph.py:71: UserWarning`
preceded by
```
        for vc in results.keys():
             add_network(net, vc)
             try:
                 sc.calc_sc(net, fault="1ph", case="max")
             except:
>                raise UserWarning("Did not converge after adding transformer with vector group %s"%vc)
E                UserWarning: Did not converge after adding transformer with vector group Yy
```

And the other errors that appear are the following:

```
        if (
            name in self._internal_names_set
            or name in self._metadata
            or name in self._accessors
        ):
            return object.__getattribute__(self, name)
        else:
            if self._info_axis._can_hold_identifiers_and_holds_name(name):
                return self[name]
>           return object.__getattribute__(self, name)
E           AttributeError: 'Series' object has no attribute 'real'

lib/python3.7/site-packages/pandas/core/generic.py:5274: AttributeError
```

The other errors that appear are in these lines:
```
lib/python3.7/site-packages/pandapower/test/shortcircuit/test_1ph.py:69: 
lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py:127: 
lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py:252: 
lib/python3.7/site-packages/pandapower/pd2ppc_zero.py:35: 
lib/python3.7/site-packages/pandapower/pd2ppc_zero.py:246: 
```

Thanks!

lthurner, 2020-03-17:
> However, I don't understand how to fix that error (which lines of code should I edit to do what) and since it is closed I thought that it was fixed already.

Yes it is fixed already, but there was no release since this was fixed. If you use the develop version on github, it should already work. But we are planning a release very soon (probably today), so if you wait it out a few hours you should be able to get this fix through pip also :)

fgfgama, 2020-03-17:
I'll probably wait, then. Thanks!

lthurner, 2020-03-17:
pandapower 2.2.2 was just released and should fix this issue. Can you confirm that it works now?

fgfgama, 2020-03-17:
Yes! Pandapower 2.2.2 successfully runs all tests!

```
================================= 442 passed, 16 skipped, 10 xfailed, 1 xpassed, 806 warnings in 325.65s (0:05:25) ==================================
```

Thank you very much @lthurner ! Now it's time to start learning how to use it.

closed 2020-03-17 by lthurner