e2nIEE/pandapower #704: Create IEEE 123-bus

MabroukAhmed, 2020-03-17:
Hi
I'm trying to model IEEE-123 bus feeder using pandapower. To create the list of network buses of this test case, I have to mention the following information:
 Bus Data Format
       1   bus number (positive integer)
       2   bus type
               PQ bus          = 1
               PV bus          = 2
               reference bus   = 3
               isolated bus    = 4
       3   Pd, real power demand (MW)
       4   Qd, reactive power demand (MVAr)
       5   Gs, shunt conductance (MW demanded at V = 1.0 p.u.)
       6   Bs, shunt susceptance (MVAr injected at V = 1.0 p.u.)
       7   area number, (positive integer)
       8   Vm, voltage magnitude (p.u.)
       9   Va, voltage angle (degrees)
   (-)     (bus name)
       10  baseKV, base voltage (kV)
       11  zone, loss zone (positive integer)
   (+) 12  maxVm, maximum voltage magnitude (p.u.)
   (+) 13  minVm, minimum voltage magnitude (p.u.)

However, In pandapower version, the only available parameters that we can mention for a given bus are: 

- name (string, default None) - the name for this bus

- index (int, default None) - Force a specified ID if it is available. If None, the index one higher than the highest already existing index is selected.

- vn_kv (float) - The grid voltage level.

- geodata ((x,y)-tuple, default None) - coordinates used for plotting

- type (string, default “b”) - Type of the bus. “n” - node, “b” - busbar, “m” - muff

- zone (string, None) - grid region

- in_service (boolean) - True for in_service or False for out of service

- max_vm_pu (float, NAN) - Maximum bus voltage in p.u. - necessary for OPF

- min_vm_pu (float, NAN) - Minimum bus voltage in p.u. - necessary for OPF

Do you have an idea how can I introduce parameters such as  Gs, shunt conductance (MW demanded at V = 1.0 p.u.) and Bs, shunt susceptance (MVAr injected at V = 1.0 p.u.) for a given bus in pandapower?
Thank you

lthurner, 2020-03-23:
This looks like a PYPOWER/MATPOWER case file structure. Are you aware pandapower has a ppc/mpc converter?

Vurgit, 2020-04-03:
Hi, have you created a ieee 123 feeder in pandapower/pypower/matpower successfully?

lthurner, 2020-05-18:
@MabroukAhmed Are you working on this? Did you have a look at the PYPOWER/MATPOWER converter?

SteffenMeinecke, 2021-11-10:
closed due to inactivity

closed 2021-11-10 by SteffenMeinecke