e2nIEE/pandapower #706: Wrong bus for plotting_toolbox.position_on_busbar() in collections.py

marhofmann, 2020-03-18:
Th bus for plotting_toolbox.position_on_busbar() in collections.create_line_switch_collection() should be 'sb' not target_bus! Otherwise it's not working...

`
intersection = position_on_busbar(net, sb, busbar_coords=line_coords)
`

marhofmann, 2020-03-18:
https://github.com/e2nIEE/pandapower/blob/50768df269aa93dd247bb8572709aff2ec934213/pandapower/plotting/collections.py#L970

lthurner, 2020-03-20:
I don't really understand this code on the first glance, didn't you yourself contribute this part? I guess you should know best then :D

Can you maybe provide a short example of a grid that plots wrong with the current implementation?

marhofmann, 2020-03-25:
Hi Leon, 

sorry for the late reply. You are right, I contributed this line and the dedicated function by myself! But since the 'position_on_busbar' has been renamed and moved to 'plotting_toolbox.py' I though somthing went wrong. I checked it with my PR and found out that the problem is sitting right in front of my screen :-D
I can't figure out why i didn't notice it earlier since i tested it quite well before pushing... Anyway - I am very sorry about that!
But let us fix it now...

By creating a simple example i noticed, that there is no documentation for 'create_busbar_collection' [in](https://pandapower.readthedocs.io/en/v2.2.2/elements/line.html) 

I created a simple example which should illustrate the problem:

`
import pandapower as pp
import pandapower.plotting as ppp

net = pp.create_empty_network()
bus = pp.create_bus(net, vn_kv=20, geodata=(0, 0))
busbar = pp.create_bus(net, vn_kv=20, geodata=(12, 20), type='b', coords=[[(10, 20), (25, 20)]])
line = pp.create_line(net, from_bus=bus, to_bus=busbar, geodata=[(0, 0), (20, 20)], length_km=2, std_type="NA2XS2Y 1x150 RM/25 12/20 kV")
pp.create_switch(net, bus=busbar, element=line, et='l')

bc = ppp.create_bus_collection(net, buses=[bus], size=1)
bbc = ppp.create_busbar_collection(net, linewidth=1)
lc = ppp.create_line_collection(net, color='k')
sc = ppp.create_line_switch_collection(net)

ppp.draw_collections(collections=[bc, bbc, lc, sc])
`
It produces this output:

[image]

Changeing it like mentioned before and it looks like this:

[image]

marhofmann, 2020-03-25:
Hi Leon, 

sorry for the late reply. You are right, I contributed this line and the dedicated function by myself! But since the 'position_on_busbar' has been renamed and moved to 'plotting_toolbox.py' I though somthing went wrong. I checked it with my PR and found out that the problem is sitting right in front of my screen :-D
I can't figure out why i didn't notice it earlier since i tested it quite well before pushing... Anyway - I am very sorry about that!
But let us fix it now...

By creating a simple example i noticed, that there is no documentation for 'create_busbar_collection' [in](https://pandapower.readthedocs.io/en/v2.2.2/elements/line.html) 

I created a simple example which should illustrate the problem:
import pandapower as pp
import pandapower.plotting as ppp

```
import pandapower as pp
import pandapower.plotting as ppp

net = pp.create_empty_network()
bus = pp.create_bus(net, vn_kv=20, geodata=(0, 0))
busbar = pp.create_bus(net, vn_kv=20, geodata=(12, 20), type='b', coords=[[(10, 20), (25, 20)]])
line = pp.create_line(net, from_bus=bus, to_bus=busbar, geodata=[(0, 0), (12, 20)], length_km=2, std_type="NA2XS2Y 1x150 RM/25 12/20 kV")
pp.create_switch(net, bus=busbar, element=line, et='l')

bc = ppp.create_bus_collection(net, buses=[bus], size=1)
bbc = ppp.create_busbar_collection(net, linewidth=1)
lc = ppp.create_line_collection(net, color='k' )
sc = ppp.create_line_switch_collection(net, distance_to_bus=2, use_line_geodata=True)

ppp.draw_collections(collections=[bc, bbc, lc, sc])
```
Produces this output:
[image]

But the switch is dedicated to the busbar and not to the node! 
Applying the changes mentioned above produces this output:
[image]

lthurner, 2020-05-18:
@marhofmann ok then - do you want to put in a PR or should I make the fix?

closed 2020-05-27 by lthurner

marhofmann, 2020-06-10:
@lthurner thank you for fixing this and sorry for not replying!

commit 29304d05 (2020-05-27): fix busbar collection Closes #706