e2nIEE/pandapower #707: Admittance Matrix Question

hectormara, 2020-03-19:
Is there some function to retrieve the Y admittance matrix for any network? Is it the makeYbus function?

And the inverse process:
Is there a function to retrieve the branch impedances for a given Y admittance matrix?

jurasofish, 2020-03-19:
Check out this tute for getting the ybus https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

If you have a poke around at the ppc data structure that contains the ybus (pycharm debugger is good for this) there's some other handy stuff in there. 

Ybus contains more than just branches, like shunt elements. You'd have to be careful going backwards.

lthurner, 2020-03-23:
Yes this is the correct answer. There is no certain way to reconstruct a network from a Y matrix, because there is much more information in the network than in the Y matrix. Just an example: there is no way to know if a admittance of 6 Ohm in Ybus represents a line of 1km with 6 Ohm/km or 2km with 3 Ohm/km. You might be able to build a network that has this Ymatrix, but you will have to make a lot of assumptions, and there are many other networks that have the same Ybus matrix.

I hope this is answered sufficiently, if not, please reopen.

closed 2020-03-23 by lthurner

hectormara, 2020-03-23:
Thank you for your answers. Just for curious people, I think if one starts with the Y admittance matrix, the lines shunt admittances B and the transformer taps, it is possible to calculate the line impedances of the original network. In fact, the off-diagonal elements are simply calculated as y_ij = - Y_ij. While for the diagonal elements y_ii the formula is more complicated and includes the above mentioned elements.