e2nIEE/pandapower #712: simple_plolty auto opens the temp_plot.html file in browser tab

zakipatel, 2020-03-22:
In a app.py file, I am running simple_plotly(net, on_map=True). 

This creates a temp_plot.html file in current directory, and auto opens the html file in a browser tab. 

Is there any way to stop this i.e. auto-opening from happening? 

I am using simple_plotly to plot a graph inside a dash application, and i cant find anything in the docs. 

Thanks !

FlorianShepherd, 2020-03-23:
No I don't think that is possible at the moment, sorry. 

But you could add the parameter if you want and pass it through to `draw_traces` where `plot()` from plotly is called. This function has the option `auto_play` which sounds like what you're looking for. Feel free to make a PR :)

jwiemer112, 2020-05-27:
For spyder users the html file doesnt open automatically. In plotly/traces.py someone changed the function _in_ipynb().

-    import __main__ as main
-    return not hasattr(main, '__file__')
+    from IPython import get_ipython
+    return get_ipython().__class__.__name__ == 'ZMQInteractiveShell'

Due to this changes the html file opens in jupyternotebooks but not in spyder. 

However there is a workarround

from pandapower.plotting.plotly import pf_res_plotly
from pandapower.networks import mv_oberrhein

net = mv_oberrhein()
fig = pf_res_plotly(net)

plot(fig)

closed 2020-11-10 by rbolgaryn