e2nIEE/pandapower #715: timeseries.run_timeseries() operation without OutputWriter

dkazacos, 2020-03-24:
Hi,

My problem: I am composing a data source that I pass to the constControl which is a dataframe with only one row that will be updated iteratively. It is the result of a series of calculations at every timestep of a simulation coming from a Prosumer model that I built and I am running. After simulating my Prosumers (loads) at every timestamp, I want to call run_times and store the results before going to the next iteration.
Trying to store some results (e.g. net.res_bus.vm_pu, ...) into a list fails to update the values after every iteration as if the data source object had constant values throughout the iteration, which it doesn't.

My solution: I found that calling OutputWriter() before the run_timeseris() and letting it output an excel file at every iteration successfully updates the results I am simultaneously storing in that empty list. So it works now!

My request: I don't want to be forced to output excel files every time I run_timeseries() in order for the results to be stored and/or successfully updated in the next iteration. I do not even need excel files at the moment. Is there any way to fix this? 

Thank you in advanced
Duarte

dkazacos, 2020-03-28:
Here is the code I am running at the moment. It is needed to iteratively run the timeseries.run_timeseries() because I need to change the mechanics of my model upon the output values dynamically
`
res = defaultdict(list)
# Need irradiante and demand at every timestep
for i, j in zip(irradiance, demand):
    # I have a bunch of houses I model
    for ind, (name, house) in enumerate(houses.items()):
        d = pd.DataFrame(index=[0])
        # Here I run my model and returns the required power from/to the grid
        load = house.run_mymodel(i, j)
        d[name] = load
        ds = ts.DFData(d)
        control.ConstControl(net, element='load',
                             element_index=ind,
                             variable='p_mw',  data_source=ds,
                             profile_name=[key])

    # Run power flow calculation at ith timestep iteration
    ow = OutputWriter(net, time_steps=[], output_path=output_dir,
                                     output_file_type=".xls", log_variables=list())

    timeseries.run_timeseries(net, verbose=False)

    # Store line overload, voltage at buses and slack power balance
    res['th_overload'].append(net.res_line.loading_percent.tolist())
    res['vm_pu_bus'].append(net.res_bus.vm_pu.tolist())
    res['slack_p'].append(net.res_ext_grid.p_mw.tolist())
`

FlorianShepherd, 2020-04-09:
I'm not sure if I either not understood what you're doing or you're using the time series module wrong. Since you're saving only the result of the last time step (this is what you're doing with `append(net.res_...)` you can just replace timeseries.run_timeseries()  with pp.runpp(net) and get the same result. In this case you don't need the outputwriter + the const controller. 

You just need to replace the ConstControl part with something like:

`net.load.at[ind, "p_mw"] = load` and set the value for each load according to your house model

dkazacos, 2020-04-16:
I see. You are completely right. I misunderstood the purpose of `run_timeseries()`. I was falling into a conceptual mistake.

Thank you very much for your response and help! I find your library very useful

closed 2020-05-18 by FlorianShepherd