e2nIEE/pandapower #718: PowerModels vs internal OPF

CollapsedDom, 2020-03-25:
There is a slight difference in the calculation method between the internal pandapower OPF calculation and PowerModels with default configuration. 
The following example only converges for the internal OPF, but not with PowerModels.
```python
import pandapower as pp
net = pp.create_empty_network()
b1 = pp.create_bus(net, vn_kv=20., max_vm_pu=1.1, min_vm_pu=0.9)
b2 = pp.create_bus(net, vn_kv=.4, max_vm_pu=1.1, min_vm_pu=0.9)
pp.create_transformer(net, b1, b2, std_type="0.4 MVA 20/0.4 kV", max_loading_percent=80)
# net.trafo['shift_degree'] = 0
ext = pp.create_ext_grid(net, b1, min_p_mw=-0.4, max_p_mw=0.4, min_q_mvar=-0.4, max_q_mvar=0.4)
pp.create_load(net, b2, p_mw=0.01, controllable=False)

# Run OPF pandapower
pp.runopp(net)

# Run OPF PowerModels
pp.runpm_ac_opf(net)
```
Setting `shift_degree` of the transformer to `0` instead of `150` according to the [std_type](https://pandapower.readthedocs.io/en/v2.2.2/std_types/basic.html#transformers), both OPF yield the same result.
I am not sure whether this is intended, but it might be a good idea to give a quick hint to that issue in the [document](https://pandapower.readthedocs.io/en/v2.2.2/opf/powermodels.html) or the [PowerModels Tutorial](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/opf_powermodels.ipynb).

lthurner, 2020-03-25:
This seems to be a PowerModels issue, there is already a ticket about this on the PowerModels issue board: https://github.com/lanl-ansi/PowerModels.jl/issues/422. A hint in the documentation would definetely be a good idea. Do you want to put in a PR for this?

CollapsedDom, 2020-03-25:
I just added another observation to the description above at the time you responded. 
It seems that the pandapower OPF by default sets `calculate_voltage_angles=False`. When enabling it, OPF does not converge as well (similar to PowerModels). Hence, I simply messed up with the default value of `calculate_voltage_angles=`, which is `True` for PowerModels call ([see](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/runpm.py#L158))and `False` for internal OPF ([see](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/run.py#L270)).

jhmenke, 2020-04-09:
>Is it intended to have different default parameters for OPF?

@FlorianShepherd   do you know this?

FlorianShepherd, 2020-04-09:
> I just added another observation to the description above at the time you responded.
> It seems that the pandapower OPF by default sets `calculate_voltage_angles=False`. When enabling it, OPF does not converge as well (similar to PowerModels). Hence, I simply messed up with the default value of `calculate_voltage_angles=`, which is `True` for PowerModels call ([see](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/runpm.py#L158)) and `False` for internal OPF ([see](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/run.py#L270)).
> 
> Is it intended to have different default parameters for OPF?

Thank you for highlighting this. No, it's not the intended behaviour. I'd suggest changing the default parameters so that both match. I think it would be best to set it to True for both.

About the angles: PowerModels doesn't support angles larger than 60 degree as said in their documentation

What @lthurner mentioned is another issue I think with the DC OPF. Here, the tap changer values are also not considered in the "DCPPowerModel". That's why we added a model called "DCMPPowerModel", which imitates the behaviour of the matpower DC OPF (which is also used in pandapower/pypower).

commit ab465f0d (2020-06-02): fixed #718
commit eb8f45f0 (2020-06-02): Merge pull request #775 from FlorianShepherd/develop

fixed #718 default runopp() OPF calculate_voltage_angles = True
fixed #792 docs example
fixes OPF tests
updated PowerModels.jl call functions to newest version
closed 2020-06-02 by FlorianShepherd