e2nIEE/pandapower #719: pf_res_plotly() for a specific time step

TiagoTorres1997, 2020-03-25:
Hello,
I was wondering how i can use the function pf_res_plotly(net) to reproduce the line loadings and bus voltages  at a specific time step?

Normally, when i apply this function it returns the plotting of the last results ( results of the time step 8759 - in my case).

For ex: at time step 4460 ( total: 8759), I observe the maximum line loading of all the lines.  How i can use pf_res_plotly() to reproduce the line loadings and bus voltages at this time step?

With gratitude,
Tiago

nailend, 2020-04-02:
Hey Tiago,

I think you will have to run a powerflow for this certain timestep and then apply the plottly function.

`apply_absolute_values(net, profiles_absolute_values, 4460)`
`pp.runpp(net)`
`pp.pf_res_plotly() `

Therefore I assume, you will also have to identify the timestep with the most 'critical' line loadings yourself. Tell me if you find an alternative solution.

regards

TiagoTorres1997, 2020-04-02:
@nailend  Thank you in advance for your answer and help!

Although, I would appreciate if you could clarify me the following doubts regarding this function: 
```ruby
apply_absolute_values(net, profiles_absolute_values, 4460)
´´´
I can't understand with this function how I can pass the results from time step 4460 to the network... The function itself `apply_absolute_values()` is not defined in the pandapower and the variable `profiles_absolut_values` should represent the network results for the time step 4460, but i don't know how to define it.

My question is: before I run the function `apply_absolute_values(net, profiles_absolute_values, 4460)`, I have to create this function to be able to use it, as well as I have to create the variable `profiles_absolute_values`. How should I do this?

Thank you again for your time and I look forward to an answer!
Best regards!

nailend, 2020-04-02:
You are right. Sorry about that. It's a function, I found in the [Simbench tutorial](https://github.com/e2nIEE/simbench/blob/develop/tutorials/simbench_grids_basics_and_usage.ipynb):

```python
def apply_absolute_values(net, absolute_values_dict, case_or_time_step):
    for elm_param in absolute_values_dict.keys():
        if absolute_values_dict[elm_param].shape[1]: # check if element is not empty
            elm = elm_param[0] # get element
            param = elm_param[1] # get parameter of element
            net[elm].loc[:, param] = absolute_values_dict[elm_param].loc[case_or_time_step] # assign parameter value of timesteps to element
```
This function might not really help you if you do not work with this dataset and therefore do not have `profiles_absolute_values` / `absolute_values_dict`. Its a dictionarry containing several dataframes with the timeseries profiles of the loads, generators, storages.

All the function does, is assigning the parameter ( 'p_mw',  'q_mvar') to the elements ('load', 'gen' etc.)
The instance of a pandapower grid can only contain the information of one timestep. I am sure you can modify this function, to suit your needs.

closed 2020-05-18 by lthurner

Platinnum, 2021-07-12:
Hello, 
even though it's been a while.... Did you manage to modify the function accordingly? Did you find a solution?
I am facing the same problem right now and wonder why there is no such functionality by default....