e2nIEE/pandapower #722: consistency with json functions

ascheidl, 2020-03-28:
to serialize a grid to a json string the function to_json is used, however to deserialize the function from_json_string is used ... we should either introduce to_json_string or remove from_json_string and include this functionality in from_json

KS-HTK, 2024-10-04:
Additionaly if this is not changed, atleast the function `from_json_string` should be documented. The docs have no mention of this function.

vogt31337, 2025-04-03:
this is available afaik since, sometimes the "data" is not coming from a file. And the from_json function reads a file to a string and calls from_json_string to convert it.
But totally we should at least provide some docstring for this function.