e2nIEE/pandapower #729: Result-reproducibility function

lschmelting, 2020-04-01:
As a first step to react to the ongoing [Replication crisis](https://en.wikipedia.org/wiki/Replication_crisis), I would like to add a function to pandapower. It's purpose is to write detailed information about the current environment to a file. I suggest to create a (hidden) directory for the file to be saved in.

Questions:
1. Name of the function (e.g.: create_metadata(), create_env_data() )?
2. Parameters?
3. Output-file type?
4. Integration with Outputwriter class via a parameter (e.g.: log_metadata or log_env_data) ?
5. Which environments should be supported? (pyenv virtualenv, conda)?

The information itself could contain:
- for example pip packages + conda packages and their versions
- System information (e.g. Architecture, Operating System, CPU info)

I'm looking forward to your suggestions.

friederikemeier, 2020-04-01:
It's important to add the commit of pandapower and possible other repos (my personal project repos). The location of the file should be given to the function as a parameter. We could add a deafult value later. If it is called from the Output writer, the directory is the outputwriter directory.

lschmelting, 2020-04-07:
After some discussion it was decided that such a function, while convient, should not be included into the output writer and pandapower, since this is a more general problem not specific to pandapower.

closed 2020-04-07 by lschmelting