e2nIEE/pandapower #736: TypeError: ufunc 'isnan' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe'' create.py, line 2385

zainistan, 2020-04-08:
Hi everyone, I am new here.

Well, I was trying to create a transformer from parameters for 3-phase power flow and I received this error. The line mentioned in the error is: 

and isnan(mag0_rx)and isnan(si0_hv_partial)and isnan(vector_group)):

I looked it up online and I think there is a conflict between the datatype of the vector_group parameter (string) and the isnan function.

I am new to Python so hopefully someone more experienced than me can look into this. Thanks!

lthurner, 2020-04-09:
Which version of pandas are you using?

zainistan, 2020-04-09:
> Which version of pandas are you using?

I am using pandas version 1.0.3.

lthurner, 2020-05-05:
Sorry for the late reply, I lost track of this issue - try downgrading to < 1.0, pandas 1.0 is pretty new and we have not yet updated support for this

commit 8b3cf8a9 (2020-06-03): fixes #736
closed 2020-06-03 by FlorianShepherd