e2nIEE/pandapower #737: Genetic algorithm -  topology that minimizes  ENS

TiagoTorres1997, 2020-04-09:
Hi,

I'm trying to develop a genetic algorithm to find the ideal network topology that minimizes the energy not supplied (ENS).  I´m kinda new at these optimisation techniques  and I was wondering if anyone can help me :)

I´m currently working with  MV Oberrhein network  and my question remains at these  points:

-  the grid operate as a radial network with 6 open sectioning points  and 316 closed, and I want to find the optimal configuration of the grid ( only changing the configuration with the 6 open, the others remain the same) that minimizes the energy not supplied;

- at this point, I want to  overload 3 or 4 lines so I can take them out of service and then run the genetic algorithm  in order to find the grid topology with lower value of ENS. 

- So after taking the lines out of service, I understood that I should elaborate some algorithm with this t "topics":
          fitness function: ENS;

          particles containing the control variables, being each variable: open or closed switch ( for only the 
          6 open switches). The   possible configurations are only regarding the 6 open switches;

          I must run multiple power flow in each iteration and compare the actual solution with the previous 
          solution; 

         The algorithm will end when the optimal grid topology, that gives me the minimum value of not 
         supplied loads, is located!

I could do this in an easier way, which was to test all combinations (turning on and off the 6 switches)  and running power flows. However, I would like to optimize the problem in question for my thesis.

I'm having a lot of difficulties understanding how to create the genetic algorithm in question, and I feel I need some help...

I apologize in advance for asking something that requires some time, and I look forward to an answer!

Best regards,
Tiago

jhmenke, 2020-04-09:
Regarding the fitness function:

You can map the binary vector of the genetic algorithm (the individual, e.g. [0 0 1 0 1 1]) to the switches by using:

```
six_oberrhein_switch_indices = net.switch.loc[net.switch.open].index  # do this once before the algorithm starts
net.switch.loc[six_oberrhein_switch_indices, "closed"] = individual
```

and then do a power flow calculation based on your overloaded grid state that you want to get the optimal topology for. You then return the energy not supplied as your cost function (or multiply by -1 to get a fitness function for maximization)

TiagoTorres1997, 2020-04-10:
@jhmenke Thank you in advance for your answer and help!

With the suggestions you gave me I went to do some more research on the subject and I still have a hard time understanding how to implement the algorithm in question.

I understand more or less the steps I have to perform , but my problem is to be able to pass the ideas to a coherent and complete script.

Being new to this kind of issues, I realised that it can be something complex to implement and I've come to ask for a little more help :)

Thank you for your time and opinion on the subject.
I wish you a blessed Easter and I look forward to an answer!

Best regards,

jurasofish, 2020-04-10:
If you link to a repository with what you've got so far I can give you some pointers.

jhmenke, 2020-04-14:
>With the suggestions you gave me I went to do some more research on the subject and I still have a hard time understanding how to implement the algorithm in question.

I'm assuming you already have a genetic algorithm and just need your fitness function, right?

TiagoTorres1997, 2020-04-14:
Unfortunately not @jhmenke...

I'm still gathering information to figure out how to implement a viable algorithm for my problem. I'm doing some research and trying to get inspired by open github repos.

If you can give me some tips and/or your opinion on how you think I should approach this problem, I'd appreciate it :)

Once again, thank you for your time and I look forward to an answer!

jhmenke, 2020-04-15:
You can take a look at DEAP, which is a evolutionary algorithm library: https://deap.readthedocs.io/en/master/

This is now out of scope since it's not a pandapower issue, so i will close it for the time being.

closed 2020-04-15 by jhmenke