e2nIEE/pandapower #739: from_json / to_json does not throw exceptions

jhmenke, 2020-04-09:
Suppose i save my file {"x":3} as "net.json".

Now i can use pp.from_json("net.json") without an error despite it not being a pandapower network. It goes through the decoder without errors. There should be a check to see if the loaded data is a valid pandapowerNet.

On the other hand, one can write pp.to_json("net.json") without actually supplying a pandapower grid. A basic type check of the arguments and a check if all arguments were supplied would solve this easily.

(Roman, i think you implemented the json encoder/decoder? can you take a quick look)?

lthurner, 2020-04-09:
> (Roman, i think you implemented the json encoder/decoder? can you take a quick look)?

I don't think this is a encode/decoder issue. The type checks should be in the from_json/to_json function before it gets handed to the encoder/decoder

jhmenke, 2020-04-09:
Wouldn't afterwards be better? Then the loaded json is a Python dict that can be validated more easily.

lthurner, 2020-04-09:
Yes that's right. So for from_json afterwards and for for_json before, but in any case in the main function and not inside the encoder/decoder...

rbolgaryn, 2020-10-14:
I do not understand why we need to throw an exception if we are saving something else that pandapower net, if the result is correct (same as normal json.dump). What if the user has a dictionary with one or more values being pandapower net, and some other values being meta-information?

jhmenke, 2020-10-19:
> I do not understand why we need to throw an exception if we are saving something else than pandapower net, if the result is correct (same as normal json.dump). What if the user has a dictionary with one or more values being pandapower net, and some other values being meta-information?

Couldn't they use the encoder directly for such cases?
In a pandapower network load function, i would expect the result to be a validated pandapower grid.

lthurner, 2020-10-19:
Agreed.

KS-HTK, 2024-11-21:
Current status:

`to_json` will dump any object to a json if possible, no type check implemented.
`from_json` will load any object but will fail as it tries to call the `convert_format` function that tries to read attributes `version` and `format_version` and will fail with error that dict has no attribute version if the passed object is not a `pandapowerNet`.

I would suggest implementing type checking and only calling `convert_format` if the loaded object is `pandapowerNet`. Let `from_json` return any other object if it is not a valid `pandapowerNet`. Same with `to_json`, why not let them encode/decode? Since these functions are not attributes of a net I do not see why they only should work for `pandapowerNet`.

Question: When is the dict converted to pandapowerNet? What keys does it need for this to happen.

## Code for testing:
```python
import pandapower as pp

net = {
  "version": "3.0.0",
  "format_version": "1"
}
pp.to_json(net, "test.json")
net = pp.from_json("test.json")

# AttributeError: 'dict' object has no attribute 'version'
```

vogt31337, 2025-04-03:
I think we could implement some kind of schema validation and use some high level tools to check that certain stuff is at least available in the net.json.

KS-HTK, 2025-10-30:
Related to #2788

closed 2026-11-10 by KS-HTK