e2nIEE/pandapower #740: Voltage level correction factor in _calc_r_x_from_dataframe

ZdenekDolezal, 2020-04-09:
When using Pandapower for running short circuits calculations I have encountered in routine "_calc_r_x_from_dataframe" ( build_branch.py ) a formula, that seems a bit unclear to me.

1) By comparison with routines "_calc_line_parameter" and "_calc_trafo_parameter" it seems probable that routine "_calc_r_x_from_dataframe" calculates pu values. In code these pu values are stored in variables z_sc, r_sc, x_sc.

2) If, for instance,  z_sc is selected, then the calculation in routine "_calc_r_x_from_dataframe" is following :
z_sc = vk_percent / 100. / sn_trafo_mva * tap_lv = 
= vk_percent / 100. / sn_trafo_mva * vn_trafo_lv**2 / vn_lv**2 * sn_mva

3) If z_sc really represents pu value, what is the meaning of voltage correction "vn_trafo_lv**2 / vn_lv**2" ( for short circuit calculations of unloaded net ).

4) Representing z_sc as pu :
a) Value pu referred to vn_trafo_lv and sn_mva
Z_nom = vn_trafo_lv**2 / sn_mva
Z_Ohm = vk_percent / 100.0 * vn_trafo_lv**2 / sn_trafo_mva
Z_pu = Z_Ohm / Z_nom = vk_percent / 100.0 / sn_trafo_mva * sn_mva
b) Value pu referred to vn_lv and sn_mva
Z_nom = vn_lv**2 / sn_mva
Z_Ohm = vk_percent / 100.0 * vn_lv**2 / sn_trafo_mva
Z_pu = Z_Ohm / Z_nom = vk_percent / 100.0 / sn_trafo_mva * sn_mva

Summary : Why the calculations in "_calc_r_x_from_dataframe" include voltage level correction factor "vn_trafo_lv**2 / vn_lv**2" ?

Thank You in advance for answer.

lthurner, 2020-04-14:
The relative values given for transformer (for example short-circuit impedance) are always relative to the nominal values of the transformer, specifically the nominal power of the transformer (trafo.sn_mva) and the nominal voltages (trafo.vn_lv_kv & trafo.vn_hv_kv). 

For the power system calculation, all values have to be in the per unit system of the power grid, which is defined by the reference power (net.sn_mva) and nominal bus voltages (net.bus.vn_kv). 

The factor vn_trafo_lv2 / vn_lv2 is in there to transform the values from rated transformer voltage to rated power system voltage. Often, these are the same, but sometimes they are not. At least in Germany, it is quite common to have 22/0.4 kV transformers in a 20/0.4 kV power system. In these cases, this transformation is necessary to get correct results.

ZdenekDolezal, 2020-04-15:
First : Thank You very much for answer
Second : I must apologize, I am not familiar enough with GitHub editor. For operation "second power" I have used the standard notation "asterisk asterisk 2", but ( as it may be clearly seen ) GitHub editor understands two concatenated asterisks as request to switch on/off bold typeface and at the same time both asterisks are "consumed" in process ( they are left out from text ). But the resulting postfixed "2" is clear enough and in the following text I will use this shortened notation.
Third : I must apologize again, I have repeated the calculation following Your description, but in the result obtained the voltage level correction factor is still missing. I must have overlooked something or have given wrong interpretation to something else, but the final formula is somewhat different from that used in _calc_r_x_from_dataframe. I will retrace the calculation in detail :
1) Getting physical value from percentual ( trafo level )
vk_perc at trafo.vn_lv_kv, trafo.sn_mva
Z_nom_trf = trafo.vn_lv_kv2 / trafo.sn_mva
vk_Ohm_trf = vk_perc / 100.0 * Z_nom_trf = vk_perc / 100.0 * trafo.vn_lv_kv2 / trafo.sn_mva
vk_Ohm_trf is physical ( Ohmical ) value on the trafo voltage level trafo.vn_lv_kv.
2) Transforming this physical value to the target voltage level ( from trafo level to net level )
Actual physical value is on the level trafo.vn_lv_kv, the target voltage level is net.bus.vn_kv, so the before mentioned value must be adjusted by coeficient
coef_trf_to_net =  net.bus.vn_kv2 /  trafo.vn_lv_kv2
i.e. :
vk_Ohm_net = 
=  coef_trf_to_net * vk_Ohm_trf = 
= net.bus.vn_kv2 /  trafo.vn_lv_kv2 * vk_perc / 100.0 * trafo.vn_lv_kv2 / trafo.sn_mva = 
= net.bus.vn_kv2 * vk_perc / 100.0 / trafo.sn_mva
vk_Ohm_net is physical ( Ohmical ) value on the net voltage level net.bus.vn_kv.
3) Getting pu value ( net level )
The pu value referred to net.sn_mva and  net.bus.vn_kv may be calculated.
Z_nom_net = net.bus.vn_kv2 / net.sn_mva
vk_pu = 
= vk_Ohm_net / Z_nom_net
= net.bus.vn_kv2 * vk_perc / 100.0 / trafo.sn_mva / ( net.bus.vn_kv2 / net.sn_mva ) =
= net.bus.vn_kv2 * vk_perc / 100.0 / trafo.sn_mva / net.bus.vn_kv2 * net.sn_mva =
= vk_perc / 100.0 / trafo.sn_mva * net.sn_mva
This formula differs from that used in _calc_r_x_from_dataframe by factor "vn_trafo_lv2 / vn_lv2". What is wrong with the calculation ?