e2nIEE/pandapower #747: problem importing pandapower in python 2.7.16

jnew123, 2020-04-16:
Hi,
I can't import pandapower. I was getting errors related to `collections.abc`. I managed to edit the files to get this sorted but now it is stuck in this:

```
File "/usr/local/lib/python2.7/dist-packages/pandapower/run.py", line 210
    parameters = {**locals(), **kwargs}
SyntaxError: invalid syntax
``` 

According to the documentation, Pandapower should work in Pyhton 2.7.
Any ideas how to get this sorted?

lthurner, 2020-04-20:
> According to the documentation, Pandapower should work in Pyhton 2.7.

Not sure what version of the documentation you are looking at, but Python 2 support has been dropped with pandapower 2.0, see #224. Python 2 will no longer be supported by most major packes, so you should really consider upgrading.

jnew123, 2020-04-20:
Hi, I read on the main page: [here](http://www.pandapower.org/start/)

[image]

lthurner, 2020-04-20:
Oh yes you are right - we missed this because it is not part of the pandapower documentation, but the pandapower website. Will fix asap!

lthurner, 2020-04-20:
Fixed in https://github.com/e2nIEE/pandapower-homepage/commit/f4661961ec097b0f62491e3ce2d6a7387474bae8

Sorry for the confusion and thanks for pointing out the inconsistency.

closed 2020-04-20 by lthurner