e2nIEE/pandapower #748: OPF tutorial not converging with constraint parameters

Hericles-Farias, 2020-04-17:
Hi, I can't perform the simulation of the pandapower Optimal Power Flow, from the tutorials, whenever I use the "net.trafo["max_loading_percent"] = 50", or the line restriction, I get the Not Converging OPF message, but without this line of code it works. I was using the pandapower 2.2.2, that I had installed via pip, then I changed to the version to 2.1.0, and in this specific version it also worked with the restrictio parameter.  What could I do to solve this issue in the 2.2.2 version?

Damowerko, 2020-05-05:
This issue is a duplicate of #733

TravisCao, 2020-08-17:
So is there any idea about solving this problem by now? @Damowerko

inratkaisija, 2020-08-24:
@Hericles-Farias Thanks for reporting this problem. Indeed, from version 2.1.0 to 2.2.0 the OPF bounds and settings for generators were changed. From version 2.2.0 on, limits or fixed values can be enforced. To make sure that the OPF converges, you can set the power tolerance to `delta=1e-16` and give it as an input to `runopp()`. Also see the updated tutorial.

commit 7a19826a (2020-08-24): set delta in runopp() and update tutorial; closes #748, closes #733
closed 2020-08-24 by lthurner