e2nIEE/pandapower #749: log variable with specific name in outputwriter

dlohmeier, 2020-04-17:
I just came across a peculiarity of the outputwriter. When adding an eval_name to a log variable (e.g. "maximum" for the minimum of a variable in some table), it is precisely stored as "maximum" globally in the numpy-output. As I am just a user and mainly working with the file output, this comes a little unexpected, as I thought that it just influences the name of the column that is stored in the file. The behavior is now that when adding "maximum" to res_line.loading_percent and to res_bus.vm_pu, csv-files are written for both, but they show the same values (which one depends on which "maximum" log variable was defined first).

Does it make sense to you to change this kind of behavior (i.e. in `_get_np_name` the name should be formed from table.value.eval_name instead of eval_name only)? Otherwise we should definitely add warnings here, in the documentation and upon adding a log variable. If the values are just overwritten, that is totally confusing.

dlohmeier, 2020-04-20:
@friederikemeier and @FlorianShepherd what do you think about this possibility:

```
    def _get_np_name(self, partial_args):
        table = partial_args[0]
        variable = partial_args[1]
        eval_name = partial_args[4]
        np_name = "%s.%s" % (table, variable)
        if eval_name is not None:
            np_name += ".%s" % eval_name
        return np_name
```

instead of this:

```
    def _get_np_name(self, partial_args):
        eval_name = partial_args[4]
        if eval_name is not None:
            return eval_name
        else:
            table = partial_args[0]
            variable = partial_args[1]
            return "%s.%s" % (table, variable)
```

lthurner, 2020-05-18:
@dlohmeier please provide a minimal example that reproduces the issue

dlohmeier, 2020-06-02:
This is what I mean:

	import pandapower as pp
	import pandapower.networks as nw
	import pandas as pd
	import pandapower.control as ctrl
	import pandapower.timeseries as ts
	import numpy as np
	import tempfile

	net = nw.case5()
	df = pd.DataFrame({0: [200, 300, 400, 500], 1: [400, 300, 100, 50], 2: [100, 300, 200, 100]})
	ds = ts.DFData(df.astype(np.float64))
	cc = ctrl.ConstControl(net, "load", "p_mw", net.load.index, profile_name=net.load.index,
						   data_source=ds)
	ow = ts.OutputWriter(net, output_path=None)
	ow.log_variable("res_sgen", "p_mw", None, np.max, "max")
	ow.log_variable("res_load", "p_mw", None, np.max, "max")
        ow.remove_log_variable("res_bus", "vm_pu")
        ow.remove_log_variable("res_line", "loading_percent")
	ts.run_timeseries(net, verbose=False)

	p_sgen = ow.output["res_sgen.p_mw"]
	p_load = ow.output["res_load.p_mw"]
	assert not np.all(p_sgen == p_load)
        assert len(ow.np_results) == 2

So, as the sgens do not have any time step values read in to them, I would expect p_sgen and p_load to be different. But as it seems, the sgen results through the "max" label are just overwritten by the load results. I think, that kind of behaviour is not really desired, as it is very implicit and does not raise any warning. Btw. both asserts fail, as the np_results only has a "max" entry. Yet this entry is copied to both "res_load.max" and "res_sgen.max" in the output.

FlorianShepherd, 2020-06-02:
Thanks @dlohmeier overwriting the results is indeed and undesired behaviour. The eval name must be unique though and I think it would be correct to raise an error if the same name is being chosen twice. I can add that

FlorianShepherd, 2020-06-02:
I added a warning in the case of the same `eval_name`. If `eval_name` is identical for two variables, the default value is now used (table + variable + index + function name).

commit 72b3be9a (2020-06-02): fixes #749 identical eval names in output writer

fixes #749 identical eval names in output writer
commit 822cae1a (2020-06-02): fixes #749
closed 2020-06-02 by FlorianShepherd