e2nIEE/pandapower #750: Clarifications of short circuit calculations and assumptions

Muda413, 2020-04-20:
To compare the short circuit study produced by another team using a commercial application, I modeled the power system with pandapower to conduct the short circuit study.

My pandapower short circuit calculation results doesn't seem to accurately reproduce their study. So seeking clarifications to ensure I'm correctly using pandapower for short circuit calculations on North American power systems.

One of such clarifications is how to model the lines and transformers since they aren't part of the standard pandapower lines and transformers library.

- For instance, #3 AWG wire, 55ft was modelled this way:
line1 = pp.create_line_from_parameters(net, from_bus=bus_3, to_bus=bus_4, length_km=0.016764, r_ohm_per_km=0.82, x_ohm_per_km=0.154, c_nf_per_km=0, max_i_ka=0.100)

How we obtained or assumed the inputs:
We obtained the r_ohm_per_km, x_ohm_per_km and max_i_ka from the NEC Code book and made c_nf_per_km = 0 since it's not a transmission line.

- For two(2) winding transformer with these parameters, we modeled it this way:
xfmr_1 = pp.create_transformer_from_parameters(net, hv_bus=0.480, lv_bus=0.208, sn_mva=0.075, vn_hv_kv=0.480, vn_lv_kv=0.208, name="xfmr_1",vk_percent=3.29, vkr_percent=0.01584, pfe_kw=0.2445, i0_percent=0.1, shift_degree=30)

How we obtained or assumed the inputs:

vkr_percent = 0.01584 - Calculated from nameplate - 0.48 * 3.3 / 100 = 0.01584

vk_percent = 3.29 - We don't have the transformer short circuit test to obtain this value. So, calculated this from the z relationship with r and x.

pfe_kw = 0.2445 kW - Interpolated iron losses based on the iron losses from an open circuit test of a known transformer in the network and their z% value

iO_percent = 0.1 - Assumed 10% because from research, smaller transformers use about 15% of rated current. 

Research can be found here - http://www.nzdl.org/gsdlmod?e=d-00000-00---off-0cdl--00-0----0-10-0---0---0direct-10---4-------0-1l--11-en-50---20-about---00-0-1-00-0--4----0-0-11-10-0utfZz-8-00&cl=CL2.5&d=HASH01ba9b176b2c8b2d5d8ef3bd.10.2&gt=1

- For 3-winding Transformer:
xfmr_3w = pp.create_transformer3w_from_parameters(net, hv_bus=16, mv_bus=0.582, lv_bus=0.582, vn_hv_kv=16.34, vn_mv_kv=0.582, vn_lv_kv=0.582, sn_hv_mva=1.645, sn_mv_mva=0.851, sn_lv_mva=0.851, vk_hv_percent=7.89, vk_mv_percent=7.99, vk_lv_percent=7.99, vkr_hv_percent=1.3072, vkr_mv_percent=0.04656, vkr_lv_percent=0.04656, pfe_kw=3.7, i0_percent=0.56, name="xfmr_3w")

How we obtained or assumed the inputs:

vk_hv_percent = 7.89 - We don't have the transformer short circuit test to obtain this value. Calculated this from the z relationship with r and x.

vk_mv_percent = 7.99 - We don't have the transformer short circuit test to obtain this value. Calculated this from the z relationship with r and x.

vk_lv_percent = 7.99 - We don't have the transformer short circuit test to obtain this value. Calculated this from the z relationship with r and x.

vkr_hv_percent = 1.3072 - used z%=8 from transformer nameplate - 16.34 * 8.00/ 100

vkr_mv_percent = 0.04656 - used z%=8 from transformer nameplate - 0.582 * 8.00/ 100

vkr_lv_percent = 0.04656 - used z%=8 from transformer nameplate - 0.582 * 8.00/ 100

- For motors used by exhaust fans:
pp.create_sgen(net, bus=ex_fan_load_bus, sn_mva=0.001119, p_mw=0.001063, q_mvar=0.000056, rx=0.5, k=0.25, type="motor", name="ex_fan")

How we obtained or assumed the inputs:
k = 1/4 = 0.25 - assuming 4 * motor FLA

sn_mva = 1.5*746W = 1119w - From motor nameplate of 1.5 hp

p_mw = 0.95*1119 = 1063w - Applying power factor of 0.95

q_mvar = 56w

rx=0.5 - A 1.5 hp motor has an x/r ~ 2; therefore rx=0.5. 
Obtained from From ANSI Standard C37.010 x/r ratio chart for induction motors on TABLE 8C which could be found here: https://arcadvisor.com/faq/motor-transformer-xr-data 

Follow up questions:

Is there any tutorial or repo you can point me to clarify all these? 

Are these the right ways to model these or what do you recommend?

Another issue was that no short circuit current value(Shows NaN) for most of the lines except the lines connected to a motor load? 
 
What do you recommend to input when there aren't any of such open and closed circuit transformer test reports?

Thank you for your usual timely response.

Muda413, 2020-04-20:
> To compare the short circuit study produced by another team using a commercial application, I modeled the power system with pandapower to conduct the short circuit study.
> 
> My pandapower short circuit calculation results doesn't seem to accurately reproduce their study. So seeking clarifications to ensure I'm correctly using pandapower for short circuit calculations on North American power systems.
> 
> One of such clarifications is how to model the lines and transformers since they aren't part of the standard pandapower lines and transformers library.
> 
> **- For instance, #3 AWG wire, 55ft was modelled this way:**
> line1 = pp.create_line_from_parameters(net, from_bus=bus_3, to_bus=bus_4, length_km=0.016764, r_ohm_per_km=0.82, x_ohm_per_km=0.154, c_nf_per_km=0, max_i_ka=0.100)
> 
> How we obtained or assumed the inputs:
> We obtained the r_ohm_per_km, x_ohm_per_km and max_i_ka from the NEC Code book and made c_nf_per_km = 0 since it's not a transmission line.
> 
> **- For two(2) winding transformer with these parameters, we modeled it this way:**
> xfmr_1 = pp.create_transformer_from_parameters(net, hv_bus=0.480, lv_bus=0.208, sn_mva=0.075, vn_hv_kv=0.480, vn_lv_kv=0.208, name="xfmr_1",vk_percent=3.29, vkr_percent=0.01584, pfe_kw=0.2445, i0_percent=0.1, shift_degree=30)
> 
> **How we obtained or assumed the inputs:**
> 
> **vkr_percent** = 0.01584 - Calculated from nameplate - 0.48 * 3.3 / 100 = 0.01584
> 
> **vk_percent** = 3.29 - We don't have the transformer short circuit test to obtain this value. So, calculated this from the z relationship with r and x.
> 
> **pfe_kw** = 0.2445 kW - Interpolated iron losses based on the iron losses from an open circuit test of a known transformer in the network and their z% value
> 
> **iO_percent** = 0.1 - Assumed 10% because from research, smaller transformers use about 15% of rated current.
> 
> Assumption based on here - http://www.nzdl.org/gsdlmod?e=d-00000-00---off-0cdl--00-0----0-10-0---0---0direct-10---4-------0-1l--11-en-50---20-about---00-0-1-00-0--4----0-0-11-10-0utfZz-8-00&cl=CL2.5&d=HASH01ba9b176b2c8b2d5d8ef3bd.10.2&gt=1
> 
> **- For 3-winding Transformer:**
> xfmr_3w = pp.create_transformer3w_from_parameters(net, hv_bus=16, mv_bus=0.582, lv_bus=0.582, vn_hv_kv=16.34, vn_mv_kv=0.582, vn_lv_kv=0.582, sn_hv_mva=1.645, sn_mv_mva=0.851, sn_lv_mva=0.851, vk_hv_percent=7.89, vk_mv_percent=7.99, vk_lv_percent=7.99, vkr_hv_percent=1.3072, vkr_mv_percent=0.04656, vkr_lv_percent=0.04656, pfe_kw=3.7, i0_percent=0.56, name="xfmr_3w")
> 
> **How we obtained or assumed the inputs:**
> 
> **vk_hv_percent** = 7.89 - We don't have the transformer short circuit test to obtain this value. Calculated this from the z relationship with r and x.
> 
> **vk_mv_percent**= 7.99 - We don't have the transformer short circuit test to obtain this value. Calculated this from the z relationship with r and x.
> 
> **vk_lv_percent** = 7.99 - We don't have the transformer short circuit test to obtain this value. Calculated this from the z relationship with r and x.
> 
> **vkr_hv_percent** = 1.3072 - used z%=8 from transformer nameplate - 16.34 * 8.00/ 100
> 
> **vkr_mv_percent** = 0.04656 - used z%=8 from transformer nameplate - 0.582 * 8.00/ 100
> 
> **vkr_lv_percent** = 0.04656 - used z%=8 from transformer nameplate - 0.582 * 8.00/ 100
> 
> **- For motors used by exhaust fans:**
> pp.create_sgen(net, bus=ex_fan_load_bus, sn_mva=0.001119, p_mw=0.001063, q_mvar=0.000056, rx=0.5, k=0.25, type="motor", name="ex_fan")
> 
> **How we obtained or assumed the inputs:**
> 
> **k** = 1/4 = 0.25 - assuming 4 * motor FLA
> 
> **sn_mva** = 1.5*746W = 1119w - From motor nameplate of 1.5 hp
> 
> **p_mw** = 0.95*1119 = 1063w - Applying power factor of 0.95
> 
> **q_mvar** = 56w
> 
> **rx**=0.5 - A 1.5 hp motor has an x/r ~ 2; therefore rx=0.5.
> Obtained from From ANSI Standard C37.010 x/r ratio chart for induction motors on TABLE 8C which could be found here: [https://arcadvisor.com/faq/motor-transformer-xr-data ](https://arcadvisor.com/faq/motor-transformer-xr-data%C2%A0)
> 
> **Follow up questions:**
> 
> Is there any tutorial or repo you can point me to clarify all these?
> 
> Are these the right ways to model these or what do you recommend?
> 
> Another issue was that no short circuit current value(Shows NaN) for most of the lines except the lines connected to a motor load?
>  
> What do you recommend to input when there aren't any of such open and closed circuit transformer test reports?
> 
> Thank you for your usual timely response.

Muda413, 2020-04-20:
Closing because I didn't add a label to it

Muda413, 2020-04-28:
@lthurner , can you help with this?

lthurner, 2020-04-28:
Regarding these questions:

>Is there any tutorial or repo you can point me to clarify all these?
>Are these the right ways to model these or what do you recommend?
>What do you recommend to input when there aren't any of such open and closed circuit transformer test reports?

I can only say that pandapower implements the short-circuit calculation according to IEC 60909 and we use the parameters as they are defined in the standard. I know how complicated it is to transform real world problems or name plate parameters from different formats, but that will have to remain up to the user. Unfortunately, I don't have the time to go through all your calculations and check them. This is just out of the scope of pandapower support.

Some general notes from going through what you wrote:
- You said you are comparing this to a short-circuit calculation from a commercial tool. In this case, I am surprised how much assumptions you had to make from literature for the input parameters. Don't you have to define them also in the commerical tool? 
- Three winding transformers are frequently a source for errors, because the short-circuit voltages can be interpreted differently. Read the note in the blue box in the pandapower documentation to understand what I mean: https://pandapower.readthedocs.io/en/v2.2.2/elements/trafo3w.html#electric-model
- It seems like you are working with a very complex network for the validation. If you want to find the reasons for the deviation, I think you will have to work with simpler minimal examples for validation. It will be very difficult to track down any mistakes in this complex network.

> Another issue was that no short circuit current value(Shows NaN) for most of the lines except the lines connected to a motor load?

Really not sure what you mean, can you make a minimal example that demonstrates this?

Muda413, 2020-04-29:
Thank you for the pointers.

Responses to some of your questions:

Don't you have to define them also in the commerical tool?
- I didn't model it myself with the commercial tool. I received a report of the short circuit studies

Here's the minimal example I could pull off to explain the NaN value I was getting:

```
def ss_network():
    
    net = pp.create_empty_network(name="net",f_hz=60.0)
  
    bus_1 = pp.create_bus(net,16.34,name="bus_1")
   
    xfmr_3_mv_bus = pp.create_bus(net,0.582,name="xfmr_3_mv_bus")
    
    xfmr_3_lv_bus = pp.create_bus(net,0.582,name="xfmr_3_lv_bus")
    
    bus_2 = pp.create_bus(net,0.480,name="bus_2")
    
    bus_3 = pp.create_bus(net,0.208,name="bus_3")
    
    bus_4 = pp.create_bus(net,0.480,name="bus_4")

    bus_5 = pp.create_bus(net,0.208,name="bus_5")
    
    bus_6 = pp.create_bus(net,0.208,name="bus_6")
    
    bus_7 = pp.create_bus(net,0.208,name="bus_7")
    
    bus_8 = pp.create_bus(net,0.48,name="bus_8")

    bus_9 = pp.create_bus(net,0.480,name="bus_9")
    
    bus_10 = pp.create_bus(net,0.480,name="bus_10")
    
    bus_11 = pp.create_bus(net,0.480,name="bus_11")
    
    bus_12 = pp.create_bus(net,0.480,name="bus_12")
    
    bus_13 = pp.create_bus(net,0.480,name="bus_13")

    bus_14 = pp.create_bus(net,0.480,name="bus_14")
    
    bus_15 = pp.create_bus(net,0.480,name="bus_15")
    
    bus_16 = pp.create_bus(net,0.480,name="bus_16")
    
    bus_17 = pp.create_bus(net,0.480,name="bus_17")
    
    pp.create_ext_grid(net,bus_1,s_sc_max_mva=0.016,rx_max=0.032)
    
    xfmr_3w = pp.create_transformer3w_from_parameters(net, hv_bus=bus_1, mv_bus=xfmr_3_mv_bus, lv_bus=xfmr_3_mv_bus, vn_hv_kv=16.34, vn_mv_kv=0.582, vn_lv_kv=0.582, sn_hv_mva=1.645, sn_mv_mva=0.851, sn_lv_mva=0.851, vk_hv_percent=7.89, vk_mv_percent=7.99, vk_lv_percent=7.99, vkr_hv_percent=1.3072, vkr_mv_percent=0.04656, vkr_lv_percent=0.04656, pfe_kw=3.7, i0_percent=0.56, name="xfmr_3w")
    
    xfmr_1 = pp.create_transformer_from_parameters(net, bus_2, bus_3, sn_mva=0.075, vn_hv_kv=0.480, vn_lv_kv=0.208, name="xfmr_1",vk_percent=3.29, vkr_percent=0.01584, pfe_kw=0.2445, i0_percent=0.1, shift_degree=30)
    line_1 = pp.create_line_from_parameters(net, bus_1, bus_2, length_km=0.016764, r_ohm_per_km=0.82, x_ohm_per_km=0.154, c_nf_per_km=0, max_i_ka=0.100, name="line_1")
    switch_1 = pp.create_switch(net,bus=bus_1,element=bus_2,et="b",closed=True,type="CB",name="switch_1")
    line_2 = pp.create_line_from_parameters(net, bus_3, bus_6, length_km=0.0039624, r_ohm_per_km=1.02, x_ohm_per_km=0.157, c_nf_per_km=0, max_i_ka=0.085, name="line_2")
    pnl_pp1_upstream_switch = pp.create_switch(net,bus=bus_6,element=bus_3,et="b",closed=True,type="CB",name="pnl_pp1_upstream_switch")
    
    
    
    xfmr_2 = pp.create_transformer_from_parameters(net, bus_4, bus_5, sn_mva=0.030, vn_hv_kv=0.480, vn_lv_kv=0.208, name="xfmr_2",vk_percent=4.09, vkr_percent=0.01968, pfe_kw=0.3038, i0_percent=0.20, shift_degree=30)
    line_3 = pp.create_line_from_parameters(net, bus_1, bus_4,length_km=0.012192, r_ohm_per_km=2.56, x_ohm_per_km=0.171, c_nf_per_km=0, max_i_ka=0.050, name="line_3")
    switch_2 = pp.create_switch(net,bus=bus_1,element=bus_4,et="b",closed=True,type="CB",name="switch_2")
    line_4 = pp.create_line_from_parameters(net, bus_5, bus_7,length_km=0.003048, r_ohm_per_km=0.82, x_ohm_per_km=0.154, c_nf_per_km=0, max_i_ka=0.100,name="line_4")
    switch_3 = pp.create_switch(net,bus=bus_7,element=bus_5,et="b",closed=True,type="CB",name="switch_3")

[... 21 lines cut ...]
    motor_3 = pp.create_sgen(net, bus=bus_14, sn_mva=0.001119, p_mw=0.001074, q_mvar=0.000045, rx=0.5, k=0.25, type="motor", name="motor_3")
    
    

    line_11 = pp.create_line_from_parameters(net, bus_1, bus_13, length_km=0.00762, r_ohm_per_km=6.6, x_ohm_per_km=0.177, c_nf_per_km=0, max_i_ka=0.025, name="line_11")
    switch_9 = pp.create_switch(net,bus=bus_1,element=bus_13,et="b",closed=True,type="CB", name="switch_9")
    line_12 = pp.create_line_from_parameters(net, bus_13, bus_15, length_km=0.009144, r_ohm_per_km=6.6, x_ohm_per_km=0.177, c_nf_per_km=0, max_i_ka=0.025, name="line_12")
    motor_4 = pp.create_sgen(net, bus=bus_15, sn_mva=0.001119, p_mw=0.0008952, q_mvar=0.0002238,rx=0.5, k=0.25, type="motor", name="motor_4")

    line_13 = pp.create_line_from_parameters(net, bus_1, bus_16, length_km=0.0762, r_ohm_per_km=1.61, x_ohm_per_km=0.167, c_nf_per_km=0, max_i_ka=0.065, name="line_13")
    switch_10 = pp.create_switch(net,bus=bus_1,element=bus_16,et="b",closed=True,type="CB", name="switch_10")
    line_14 = pp.create_line_from_parameters(net, bus_16, bus_17, length_km=0.0762, r_ohm_per_km=2.56, x_ohm_per_km=0.171, c_nf_per_km=0, max_i_ka=0.050, name="line_14")
    load_1 = pp.create_load(net,bus=bus_16,p_mw=0.192, q_mvar=0.0096, name="load_1")
    switch_11 = pp.create_switch(net,bus=bus_17,element=line_14,et="l",closed=True,type="CB", name="switch_11")
    
    
    return net
```

When I run this command below, only lines 10,12 and 14 return values while others return NaN

``` 
sc.calc_sc(net,fault="3ph",case="max", ip=True, ith=True, branch_results=True)
net.res_line_sc
```

lthurner, 2020-04-30:
> I didn't model it myself with the commercial tool. I received a report of the short circuit studies

If you don't have the original data sets, I think you will have a very hard time reproducing or validating the results. In my personal experience there are just to many unknowns in this equation. You are making a lot of assumptions, and if in any of these cases the person who used the other tool made different assumptions, the results will be different. I don't see how that can match up...

Muda413, 2020-05-04:
I understand and feel you @lthurner ,but do you have an idea why I'm getting the NaN values?

lthurner, 2020-05-04:
I am looking into it, seems like the branch results are "blocked" by the transformers, the results are only on the high voltage side.

One thing that I also noticed it that your 3W transformer is connected to the same bus on LV and MV, is that intended? I don't know if thats valid (its at least not tested)

lthurner, 2020-05-04:
If you run 

> pp.diagnostic(net)

on the net that you have provided, you can see that there are several lines that connect different voltage levels, and also multiple lines that have impedance values close to zero, which could lead to problems when inverting the admittance matrix.

Can you please provide an example of a network that passes the pp.diagnostic test? If the issue still occurss, I can look into it.

Muda413, 2020-05-05:
> I am looking into it, seems like the branch results are "blocked" by the transformers, the results are only on the high voltage side. I am looking into why this is. The branch results are not yet really well tested, which is why there is also a warning when you run short-circuit calculation with branch_results.
> 
> One thing that I also noticed it that your 3W transformer is connected to the same bus on LV and MV, is that intended? I don't know if thats valid (its at least not tested). That doesn't seem to be the reason for this problem above though

Muda413, 2020-05-05:
> I am looking into it, seems like the branch results are "blocked" by the transformers, the results are only on the high voltage side. I am looking into why this is. The branch results are not yet [...]

Isn't that the right way to model for 3-winding transformers. the hv_bus is connected to the 16.3kv bus while the mv and lv buses are connected to the 0.582 kv bus

Muda413, 2020-05-05:
> > pp.diagnostic(net)

Thank you for this. It made debugging way faster and easier. The network now passes the test but the issue still occurs. Below is the updated network:

```
def ss_network():
    
    net = pp.create_empty_network(name="net",f_hz=60.0)
    
    bus_1 = pp.create_bus(net,16.34,name="bus_1")
    
    xfmr_3_mv_bus = pp.create_bus(net,0.582,name="xfmr_3_mv_bus")
    xfmr_3_lv_bus = pp.create_bus(net,0.582,name="xfmr_3_lv_bus") 
    
    
    xfmr_1_hv_bus = pp.create_bus(net,0.480,name="xfmr_1_hv_bus")
    xfmr_1_lv_bus = pp.create_bus(net,0.208,name="xfmr_1_lv_bus")

    
    xfmr_2_hv_bus = pp.create_bus(net,0.480,name="xfmr_2_hv_bus")
    xfmr_2_lv_bus = pp.create_bus(net,0.208,name="xfmr_2_lv_bus")
    
    
    xfmr_0_hv_bus = pp.create_bus(net,16.34,name="xfmr_0_hv_bus")
    xfmr_0_lv_bus = pp.create_bus(net,0.480,name="xfmr_0_hv_bus")
    
    bus_2 = pp.create_bus(net,0.480,name="bus_2")
    
    bus_3 = pp.create_bus(net,0.208,name="bus_3")
    

    bus_4 = pp.create_bus(net,0.208,name="bus_4")
    

    bus_5 = pp.create_bus(net,0.480,name="bus_5")
     

    bus_6 = pp.create_bus(net,0.480,name="bus_6")

    bus_8 = pp.create_bus(net,0.480,name="bus_8")
    
    bus_9 = pp.create_bus(net,0.480,name="bus_9")
     
    bus_11 = pp.create_bus(net,0.480,name="bus_11")
    
    bus_12 = pp.create_bus(net,0.480,name="bus_12")
    
    bus_13 = pp.create_bus(net,0.480,name="bus_13")

    bus_14 = pp.create_bus(net,0.480,name="bus_14")
    
    bus_15 = pp.create_bus(net,0.480,name="bus_15")
    
    bus_16 = pp.create_bus(net,0.480,name="bus_16")
    
    bus_17 = pp.create_bus(net,0.480,name="bus_17")
    
    
    pp.create_ext_grid(net,bus_1,s_sc_max_mva=0.016,rx_max=0.032)
    
    xfmr_3w = pp.create_transformer3w_from_parameters(net, hv_bus=bus_1, mv_bus=xfmr_3_mv_bus, lv_bus=xfmr_3_lv_bus, vn_hv_kv=16.34, vn_mv_kv=0.582, vn_lv_kv=0.582, sn_hv_mva=1.645, sn_mv_mva=0.851, sn_lv_mva=0.851, vk_hv_percent=7.89, vk_mv_percent=7.99, vk_lv_percent=7.99, vkr_hv_percent=1.3072, vkr_mv_percent=0.04656, vkr_lv_percent=0.04656, pfe_kw=3.7, i0_percent=0.56, name="xfmr_3w")
    
    
    xfmr_0 = pp.create_transformer_from_parameters(net, xfmr_0_hv_bus, xfmr_0_lv_bus, sn_mva=0.3, vn_hv_kv=16.34, vn_lv_kv=0.480, name="xfmr_0",vk_percent=5.6849, vkr_percent=0.8627, pfe_kw=0.426, i0_percent=0.28, shift_degree=30)
    line_0 = pp.create_line_from_parameters(net, bus_1, xfmr_0_hv_bus, length_km=0.001524, r_ohm_per_km=2.49, x_ohm_per_km=1.151, c_nf_per_km=0, max_i_ka=0.130, name="line_0")
    switch_0 = pp.create_switch(net,bus=bus_1,element=xfmr_0_hv_bus,et="b", closed=True, type="CB", name="switch_0")
    line_0_1 = pp.create_line_from_parameters(net, xfmr_0_lv_bus, bus_2, length_km=0.006096, r_ohm_per_km=2.062, x_ohm_per_km=1.125, c_nf_per_km=0, max_i_ka=0.475,name="line_0_1")
    switch_0_1 = pp.create_switch(net,bus=xfmr_0_lv_bus,element=bus_2,et="b",closed=True,type="CB", name="switch_0_1")
    
    
[... 28 lines cut ...]
    motor_2 = pp.create_sgen(net, bus=bus_9, sn_mva=0.00506, p_mw=0.004807, q_mvar=0.000253, rx=0.167, k=0.25, type="motor", name="motor_2")
    
   
    
    line_9 = pp.create_line_from_parameters(net, bus_2, bus_11, length_km=0.009144, r_ohm_per_km=6.6, x_ohm_per_km=0.177, c_nf_per_km=0, max_i_ka=0.025, name="line_9")
    switch_8 = pp.create_switch(net,bus=bus_2,element=bus_11,et="b",closed=True,type="CB", name="switch_8")
    line_10 = pp.create_line_from_parameters(net, bus_11, bus_12, length_km=0.01524, r_ohm_per_km=6.6, x_ohm_per_km=0.177, c_nf_per_km=0, max_i_ka=0.025, name="line_10")
    motor_3 = pp.create_sgen(net, bus=bus_12, sn_mva=0.001119, p_mw=0.001074, q_mvar=0.000045, rx=0.5, k=0.25, type="motor", name="motor_3")
    
    

    line_11 = pp.create_line_from_parameters(net, bus_2, bus_13, length_km=0.00762, r_ohm_per_km=6.6, x_ohm_per_km=0.177, c_nf_per_km=0, max_i_ka=0.025, name="line_11")
    switch_9 = pp.create_switch(net,bus=bus_2,element=bus_13,et="b",closed=True,type="CB", name="switch_9")
    line_12 = pp.create_line_from_parameters(net, bus_13, bus_14, length_km=0.009144, r_ohm_per_km=6.6, x_ohm_per_km=0.177, c_nf_per_km=0, max_i_ka=0.025, name="line_12")
    motor_4 = pp.create_sgen(net, bus=bus_14, sn_mva=0.001119, p_mw=0.0008952, q_mvar=0.0002238,rx=0.5, k=0.25, type="motor", name="motor_4")

    line_13 = pp.create_line_from_parameters(net, bus_2, bus_15, length_km=0.0762, r_ohm_per_km=1.61, x_ohm_per_km=0.167, c_nf_per_km=0, max_i_ka=0.065, name="line_13")
    switch_10 = pp.create_switch(net,bus=bus_2,element=bus_15,et="b",closed=True,type="CB", name="switch_10")
    line_14 = pp.create_line_from_parameters(net, bus_15, bus_16, length_km=0.0762, r_ohm_per_km=2.56, x_ohm_per_km=0.171, c_nf_per_km=0, max_i_ka=0.050, name="line_14")
    load_1 = pp.create_load(net,bus=bus_16,p_mw=0.192, q_mvar=0.0096, name="load_1")
    switch_11 = pp.create_switch(net,bus=bus_16,element=bus_17,et="b",closed=True,type="CB", name="switch_11")
    
    
    return net
```

Muda413, 2020-05-12:
@lthurner , mistakenly closed this when posting comments last week. The NaN issue still occurs. Can you please look into it as you had suggested previously

Muda413, 2020-05-19:
@miek770, I observed you have some experience with using pandapaower for North American short circuit studies. Anyway you can take a stab at this challenges we are having here?

lthurner, 2020-06-16:
Just looked into it now: the problem is with the bus-bus switches you have in the net. 

>     line_6 = pp.create_line_from_parameters(net, bus_5, bus_6, length_km=0.003048, r_ohm_per_km=2.82, x_ohm_per_km=1.154, c_nf_per_km=0, max_i_ka=0.100, name="line_6")
> switch_5 = pp.create_switch(net,bus=bus_5,element=bus_6,et="b",closed=True,type="DS",name="switch_5")

This is a line and switch that connects the same two buses. The switch is modeled as having zero impedance, so you are basicly short-wiring these two buses. Its only logical that the line that connects these two buses carries no current, neither in power flow nor in short-circuit.

closed 2020-06-18 by lthurner