e2nIEE/pandapower #754: Ext Grid Appearance in SimplePlot

SimonRubenDrauz, 2020-04-21:
if using simple_plot.py the ext_grid is plotted as yellow square. I would suggest to take a hatched ext_grid instead, as ext_grids are also drawn hatched in PowerFactory and Sincal. 

The changes have been pushed to my fork under following branch:
https://github.com/SimonRubenDrauz/pandapower/tree/hatch

While adapting that, I realized another thing: creating a ext_grid_collection in simple_plot.py calls create_bus_collection, instead of create_ext_grid_collection. Maybe we should adapt this too.

FlorianShepherd, 2020-06-02:
Would be fine for me and is a good idea I guess. Thanks @SimonRubenDrauz  Would you create a PR?

SteffenMeinecke, 2021-11-10:
@SimonRubenDrauz how is the status? Is that included in https://github.com/e2nIEE/pandapower/pull/1378 ?

rbolgaryn, 2022-02-04:
This is the current state:

[image]

So closing the issue

closed 2022-02-04 by rbolgaryn