e2nIEE/pandapower #759: Drop not required lines when fusing two buses

dlohmeier, 2020-04-28:
I just came across another interesting aspect in one of the toolbox-functions. When calling `fuse_buses`, all elements that are connected to b2 will be connected to b1 instead. For branch elements, this does not consider that b1 might also be standing "on the other side" of the branch, which would lead to the branch connecting b1 to itself. Interestingly the case of switches is considered here, probably because their 'et' can differ making it harder to consider them in general.

I would like to make two suggestions for this:
1. Call `drop_lines` and `drop_trafos` before changing all the bus columns from b2 to b1.
2. Consider branch_elements and node_elements separately in `bus_element_tuples` so that for node_elements only one column is returned, whereas for branch_elements a list is returned. It could look like this: {"node_elements": {"sgen": "bus", "load": "bus", ... }, "branch_elements": {"line": ("from_bus", "to_bus"), "trafo": ("hv_bus", "lv_bus"), "trafo3w": ("hv_bus", "mv_bus", "lv_bus"), ... }} .

What do you think about either of these ideas?

lthurner, 2020-05-18:
@dlohmeier please provide a minimal example as a failing test

SteffenMeinecke, 2020-05-18:
Refering point 1:
I'm working on an proposal (PR).

Refering point 2:
I don't see the need for {"node_elements": {"sgen": "bus", "load": "bus", ... }} since you can create that simply by `{key: val for key, val in pp.element_bus_tuples(branch_elements=False)}`. For the branch elements `branch_element_bus_dict()` already exists.

dlohmeier, 2020-05-18:
OK, I think this aspect is not really required. I just aproached it from the other side, but I think the PR #774 makes sense.

closed 2020-05-25 by SteffenMeinecke