e2nIEE/pandapower #761: AssertionError when running the testsuite under Python 3.8.2, alpine Linux

hristiy4n, 2020-05-01:
The following error is encountered when running the testsuite:

```cli
pandapower/test/estimation/test_opt_lp_estimation.py   NIT   NF   F                       GTG
    0    1  2.907781954177508E+07   4.32265548E+18
tnc: fscale = 2.502e-10
    1    5  1.749321423858594E+07   6.52432243E+17
    2   13  4.631323100454346E+06   4.58513459E+16
    3   18  3.001500553672809E+06   2.50006148E+16
tnc: fscale = 5.03576e-09
    4   30  1.961013517917778E+06   2.29846886E+16
    5   36  1.539278767144397E+06   2.05036270E+16
    6   64  5.562675773984008E+05   3.21265144E+15
    7   80  1.144530389716266E+05   5.17938180E+14
    8   92  4.409539773837221E+03   9.90510877E+13
    9  100  9.674743242832533E+02   8.14941207E+12
tnc: fscale = 2.25123e-07
   10  112  6.271274139265605E+02   2.97354719E+12
   11  124  5.764129692742497E+02   2.45987381E+12
   12  131  5.619094346654595E+02   8.70352960E+10
   13  141  5.530128329769578E+02   3.16326668E+10
   14  151  5.503735427200554E+02   1.66523094E+10
   15  181  5.474940226892033E+02   1.13641347E+10
tnc: fscale = 4.99096e-06
tnc: |xn-xn-1] = 5.73846e-07 -> convergence
   16  185  5.474553555024202E+02   2.73794694E+09
tnc: Converged (|x_n-x_(n-1)| ~= 0)
F

====================================================================== FAILURES ======================================================================
_____________________________________________________ test_case30_compare_classical_wls_opt_wls ______________________________________________________

    def test_case30_compare_classical_wls_opt_wls():
        net = nw.case30()
        pp.runpp(net)
        add_virtual_meas_from_loadflow(net)
    
        try:
            success = estimate(net, init='flat', algorithm="opt", estimator='wls')
            assert success
        except:
            # if failed give it a warm start
            net, ppc, eppci = pp2eppci(net)
            estimation_wls = WLSAlgorithm(1e-3, 3)
            estimation_opt = OptAlgorithm(1e-6, 1000)
    
            eppci = estimation_wls.estimate(eppci)
            eppci = estimation_opt.estimate(eppci, estimator="wls")
            assert estimation_opt.successful
            net = eppci2pp(net, ppc, eppci)
    
        net_wls = deepcopy(net)
        estimate(net_wls)
>       assert np.allclose(net_wls.res_bus_est.vm_pu, net.res_bus_est.vm_pu, atol=1e-2)
E       assert False
E        +  where False = <function allclose at 0x7fdf30d90b80>(0     0.998544\n1     0.998548\n10    0.979024\n11    0.983992\n12    0.998544\n13    0.975188\n14    0.978746\n15    0.97590...9\n4     0.980995\n5     0.971691\n6     0.965867\n7     0.959110\n8     0.979024\n9     0.982929\nName: vm_pu, dtype: float64, 0     1.027158\n1     1.027173\n10    1.008022\n11    1.012893\n12    1.027042\n13    1.004307\n14    1.007772\n15    1.00501...1\n4     1.010062\n5     1.001015\n6     0.995354\n7     0.988803\n8     1.008044\n9     1.011829\nName: vm_pu, dtype: float64, atol=0.01)
E        +    where <function allclose at 0x7fdf30d90b80> = np.allclose
E        +    and   0     0.998544\n1     0.998548\n10    0.979024\n11    0.983992\n12    0.998544\n13    0.975188\n14    0.978746\n15    0.97590...9\n4     0.980995\n5     0.971691\n6     0.965867\n7     0.959110\n8     0.979024\n9     0.982929\nName: vm_pu, dtype: float64 =        vm_pu  va_degree       p_mw     q_mvar\n0   0.998544   0.000000 -25.973966   0.996594\n1   0.998548  -0.416796 -3...33890  29.999724  29.997631\n8   0.979024  -3.005864  -0.000079   0.000036\n9   0.982929  -3.385014   5.799881   2.000122.vm_pu
E        +      where        vm_pu  va_degree       p_mw     q_mvar\n0   0.998544   0.000000 -25.973966   0.996594\n1   0.998548  -0.416796 -3...33890  29.999724  29.997631\n8   0.979024  -3.005864  -0.000079   0.000036\n9   0.982929  -3.385014   5.799881   2.000122 = This pandapower network includes the following parameter tables:\n   - bus (30 elements)\n   - load (20 elements)\n   - g...r_flow (2 elements)\n   - res_gen_power_flow (5 elements)\n   - res_bus_est (30 elements)\n   - res_line_est (41 elements).res_bus_est
E        +    and   0     1.027158\n1     1.027173\n10    1.008022\n11    1.012893\n12    1.027042\n13    1.004307\n14    1.007772\n15    1.00501...1\n4     1.010062\n5     1.001015\n6     0.995354\n7     0.988803\n8     1.008044\n9     1.011829\nName: vm_pu, dtype: float64 =        vm_pu  va_degree       p_mw     q_mvar\n0   1.027158   0.000000 -25.973445   1.147675\n1   1.027173  -0.394076 -3...78858  30.006307  30.085148\n8   1.008044  -2.836382   0.003271   0.030458\n9   1.011829  -3.194019   5.806235   2.022938.vm_pu
E        +      where        vm_pu  va_degree       p_mw     q_mvar\n0   1.027158   0.000000 -25.973445   1.147675\n1   1.027173  -0.394076 -3...78858  30.006307  30.085148\n8   1.008044  -2.836382   0.003271   0.030458\n9   1.011829  -3.194019   5.806235   2.022938 = This pandapower network includes the following parameter tables:\n   - bus (30 elements)\n   - load (20 elements)\n   - g...r_flow (2 elements)\n   - res_gen_power_flow (5 elements)\n   - res_bus_est (30 elements)\n   - res_line_est (41 elements).res_bus_est

pandapower/test/estimation/test_opt_lp_estimation.py:42: AssertionError
```
python version - 3.8.2
numpy version - 1.18.3
pandas version - 1.0.3

Running the testsuite under python 3.7.5 with numpy version 1.16.4 does not seem to result the same error.

wangzhenassd, 2020-05-02:
Hi,
what is your SciPy version?

hristiy4n, 2020-05-02:
The version of SciPy is 1.3.3. When tested with python 3.7.5 the version was 1.3.1.

wangzhenassd, 2020-05-02:
Due to the limitation of OS, I only tried your version setup on Windows, but cannot recreate the same error.  I am not sure of the reason of the failed test, maybe you can try update scipy.

lthurner, 2020-05-04:
Python 3.8 is not yet included in Travis CI, so 3.8 support is not yet tested for pandapower. Should we include it with the goal to support 3.8 in the next release?

wangzhenassd, 2020-05-07:
@lthurner I think you are right, sooner or later pandapower should be compatible with the latest python version and packages e.g. pandas.

lthurner, 2020-05-18:
We will continue in #772

closed 2020-05-18 by lthurner