e2nIEE/pandapower #762: numba and state estimation

znicoy, 2020-05-06:
Hi,
do you plan to use numba for the state estimation as well? Or am I overseeing something and it is actually already being used?
Thanks in advance!
Nico

wangzhenassd, 2020-05-07:
Hi @znicoy ,
currently there is no plan to implement Numba acceleration for state estimation over the vectorized Numpy/Scipy version, since the sparse matrix operation is much more volatile than in powerflow due to the allocation of measurement.

znicoy, 2020-05-07:
Thanks for your answer! So you think numba will not improve the speed at all in that case? And where do you see the potential to optimize the code?

wangzhenassd, 2020-05-08:
I think, the potential is mainly on pd2ppc process including the conversion of measurement. The sparse matrix operation performance is mainly limited by the SciPy library.  Is your question rightfully explained?

znicoy, 2020-05-08:
Yes, thanks!

closed 2020-05-08 by znicoy