e2nIEE/pandapower #766: State Estimation minimal example

2malto, 2020-05-08:
Hi,

i am interested in State Estimation at pandapower and started with a minimal example (see below). The minimal net includes three measurements (one bus  and two line measurements).

I integrated the network into pandapower and started the StateEstimation to get the estimate values for the absolute voltage and the voltage angle at bus „b2“. Unfortunately I struggle with the results in pandapower. So the absolute voltage at slack bus "b1" is quite low (0.94 pu) and the absolute voltage at bus „b2“ does not really deviate from the measurement value.

I expected a fixed absolute voltage value at the slack bus (1 pu) and values like 0.997 pu for absolute voltage at bus „b2“ and -18 degree  for voltage angle at bus „b2“.

I will be grateful for any help, advice or option, that I missed.

Thanks in advance and keep working on pandapower. It´s great!

Kind regards 
Malte

`net = pp.create_empty_network()`
`b1 = pp.create_bus(net, vn_kv=220)`     
`b2 = pp.create_bus(net, vn_kv=220)`
`pp.create_ext_grid(net, bus = b1, vm_pu=1)`
`l1 = pp.create_line_from_parameters(net,from_bus=b1, to_bus=b2, length_km=1, r_ohm_per_km=0, x_ohm_per_km=100, c_nf_per_km=0, max_i_ka= 1)`

`pp.create_measurement(net, meas_type= "p", element_type= "line", value= 155, std_dev= 1, element=l1, side = b1)`   
`pp.create_measurement(net, meas_type= "q", element_type= "line", value= 22, std_dev= 1, element=l1, side = b1)`    
`pp.create_measurement(net, meas_type= "v", element_type= "bus", value= 210/220, std_dev= 1/220, element=b2)`
`net.measurement`

`est.estimate(net, init= "flat", algorithm= "wls")`

Results: 
[image]

wangzhenassd, 2020-05-08:
Hi @2malto,
State estimation is slightly different from power flow on the slack voltage definition, where only the voltage angle is defined as reference, the slack voltage magnitude may vary upon your defined measurement. In your case, with the "b1" vm_pu = 0.94 p.u. results in the wished branch flow of p=155, q=22. I think, this is the reason for this behavior, which is also expected.

2malto, 2020-05-11:
Hi @wangzhenassd ,
thank you for your advice and help! I just recognize my mistake.

I just calculated the minimal example by hand. I supposed, that the slack voltage magnitude is exactly given and fixed. So, in my case the real part of the slack voltage magnitude is not a variable of the measurement model nor part of the true state vector x. That leads to results like 1pu at slack bus.

But by adding the slack voltage magnitude (real part) to the true state vector x, I get the same result as in PandaPower.

Do you know, if there is an option/possibility in PandaPower to fix the slack voltage magnitude as well? So that the slack voltage is not part of the true state vector x and doesn´t vary?

Thanks in advance and have a good day.

jhmenke, 2020-05-18:
>Do you know, if there is an option/possibility in PandaPower to fix the slack voltage magnitude as well? So that the slack voltage is not part of the true state vector x and doesn´t vary?

You could define the slack bus voltage measurement with a very low standard deviation, e.g. 1e-5, so it will most likely be fixed at the measured value (which you set to the slack bus voltage magnitude).

closed 2020-05-18 by jhmenke