e2nIEE/pandapower #771: implement distributed slacks

nailend, 2020-05-12:
#218 
The Pandapower [Website](https://www.pandapower.org/about/#analysis) states that Transmission Grids are in the framework scope. In order to calculate load flow calculations in the meshed German high-voltage grid it seems to make sense to use distributed slacks. 

- Are there already first approaches?
- Is it possible to transfer the [PyPSA](https://github.com/PyPSA/PyPSA/issues/10) approach to panda power?
- Are there huge differnces in the architecture of these 2 frameworks?

lthurner, 2020-05-18:
Hi, we currently use our controller structure to implement distributed slacks with a loop around the power flow (so not inside the newton-raphson, but running a power flow, adapting the inputs, running power flow again until it is converged).

Obviously having this inside the power flow algorithm is the more elegant solution, and that was also on our development roadmap, although we didn't get around to it yet. I wasn't aware that PyPSA had implemented this already, that is awesome! There are significant differences in the frameworks, but mostly in the data structure and API - just with a quick look, I think it should be possible to integrate this into pandapower with the inspiration of the existing implementation in PyPSA.

Are you able to help with this? If so, I could definitely point you in the right direction for where to start.

jurasofish, 2020-05-19:
This is something I could take up, if you're not already on it @nailend?

nailend, 2020-05-21:
@jurasofish no, please take the lead as I am not yet too familiar with the code.

jurasofish, 2020-05-22:
@lthurner could you point me in the right direction to get started?
My initial thoughts are to 
- Add a contribution factor column to generators and external grids (not static gens or dc lines).
- Default the contribution factor to 1 for external grids and to 0 for generators.
- Enable distributed slack through an argument to runpp (e.g. ``pp.runn(net, distributed_slack=True)``
- Implement distributed slack for Newton Raphson only. Raise exception if it is used in other load flows. Other methods are future TODOs.

I haven't looked closely at the OPF or state estimation areas of pandapower - do you expect these will need modification?

lthurner, 2020-05-25:
Thanks a lot for getting involved, much appreciated!

Your description is spot on, this is exactly the way to do it. If you need any help along the way, let me know.

> I haven't looked closely at the OPF or state estimation areas of pandapower - do you expect these will need modification?

There might be some issues if you modify the pd2ppc function, since that is used by multiple methods (OPF, short-circuit etc). But that will be easy to solve afterwards, so you don't need to worry about that. If you get some failing tests from state estimation or other methods, we can fix it afterwards.

jurasofish, 2020-06-15:
@lthurner I wanted to run by how I'm implementing this with you.
In this commit (https://github.com/jurasofish/pandapower/commit/e70535c34f8a29ba3275dc7cb84832d0cd8323ed) I've added contribution factors by expanding the ppc gen array to add a new column. I guess this sort-of might make it incompatible with ppc now since the array has an extra column. Your thoughts on this approach? 

As an alternative I could add this new column as a vector in the same datastructure as the gen array, but then I'd have to fiddle around making sure that vector follows the generators around when they're reordered, removed for being out of service (if that happens), etc. especially in the conversion to ppci.

wangzhenassd, 2020-08-11:
Hi all, Is there any progress or plan on adding extra column in pandapower network to identify the priority/weight of distributed slacks?

jurasofish, 2020-08-11:
Yeah, I've been working on it here https://github.com/e2nIEE/pandapower/compare/develop...jurasofish:dist_slack
Have added a column to external grids dataframe and to generators dataframe.
Current status is that it takes the weights, associates them with generators, normalises weights within each island to 1, and feeds the weights through into the main newtonpf function.

I've been neglecting this a bit - pretty flat out with work atm.

wangzhenassd, 2020-08-12:
Thank you! I will have a look on that.

lthurner, 2020-08-18:
@jurasofish : sorry your question about the implementation slipped through.

> @lthurner I wanted to run by how I'm implementing this with you.
>In this commit (jurasofish@e70535c) I've added contribution factors by expanding the ppc gen array to add a new column. I 
> guess this sort-of might make it incompatible with ppc now since the array has an extra column. Your thoughts on this approach?

This is a fine approach. We have already added columns for extra parameters, such as impedance loads that are not in the original data structure. This doesn't make it incompatible either, because pypower/matpower applications would just ignore the extra columns. I agree it is cleaner than always having to keep track of the sorting. 

Would be greatly appreciated if you find more time to work on this!

edxu96, 2020-09-27:
Hi, thanks for the effort on this topic. I am wondering if someone can recommend some literature or even descriptions. I'd like to include this part in my thesis. Look forward to the implementation as well.

jurasofish, 2020-09-28:
Power System Modelling and Scripting by Federico Milano (2010) has a model. Quite a terse book though.
Good luck with your thesis!

rbolgaryn, 2021-01-14:
> 
> 
> @lthurner could you point me in the right direction to get started?
> My initial thoughts are to
> 
>     * Add a contribution factor column to generators and external grids (not static gens, dc lines or extended wards).
> 
>     * Default the contribution factor to 1 for external grids and to 0 for generators.
> 
>     * Enable distributed slack through an argument to runpp (e.g. `pp.runn(net, distributed_slack=True)`
> 
>     * Implement distributed slack for Newton Raphson only. Raise exception if it is used in other load flows. Other methods are future TODOs.
> 
> 
> I haven't looked closely at the OPF or state estimation areas of pandapower - do you expect these will need modification?

Hi @jurasofish ,

Thank you for taking the lead on this!

I have noticed that having several disconnected islands in the same grid model could be a challenge for using the distributed slack. Would your implementation be able to work in such a situation?

Roman

rbolgaryn, 2021-01-15:
Also, slack buses can be modelled by gen elements that have the values True in the column "slack", which would be relevant for setting default contribution factors.

rbolgaryn, 2021-07-19:
Just opened a pull request with a working implementation of distributed slack calculation for ext_grid and gen. 

Thank you @jurasofish for the initial work for this implementation!

Roman

rbolgaryn, 2021-07-22:
The pull request is merged.
There is one open point still, implementation for separate zones.
I will close this issue and open a new one, just for that point.

For other methods apart from NR, further issues can be opened, when the need arises.

Roman

closed 2021-07-22 by rbolgaryn