e2nIEE/pandapower #777: Participate in FMI Cross Check

chrbertsch, 2020-05-19:
According to https://fmi-standard.org/tools/, Pandapower can export FMUs, but does not yet participate in the  FMI Cross Check https://github.com/modelica/fmi-cross-check. 
For many tools this has significantly improved the maturity of their FMI support.

lthurner, 2020-05-19:
Not sure exactly how we ended up on that list, at least I am not aware that pandapower exports FMUs... Might be worth looking into though, thanks for pointing that out.

AdrienGougeon, 2020-06-16:
Hello,
For my work I implemented the possibility to export FMUs from pandapower, you can see the project here: https://framagit.org/Adrien.Gougeon/pandapower-fmu
I asked to add pandapower as a validated tool, https://github.com/modelica/fmi-cross-check/pull/84, but never obtained any answer

chrbertsch, 2021-02-22:
> I asked to add pandapower as a validated tool, modelica/fmi-cross-check#84, but never obtained any answer

@t-sommer : do you have an idea why the exported pandapower-fmu FMUs do not get listed on https://fmi-standard.org/cross-check/fmi2-cs-linux64/? 
Is it, because no importing tool has reported results for these FMUs (https://github.com/AdrienGougeon/fmi-cross-check/tree/0dbd5e23bcfc31e2b9e4791488f357cb6cbe6531/results/2.0/cs/linux64)

vogt31337, 2024-11-22:
@AdrienGougeon sadly your repository does not contain any license. So maybe you could make a PR with your work and put your code in the converter section?

chrbertsch, 2024-11-22:
The "FMI Cross Check" has been replaced by the "Compatiblity information", see https://github.com/modelica/fmi-standard.org/blob/main/CONTRIBUTING.md#compatibility-information for more information