e2nIEE/pandapower #780: Intended behaviour with multiple slacks

jurasofish, 2020-05-24:
I'm currently working on #771 and have been familiarising myself with exactly how pandapower handles slack generators.
I can't find anything in the documentation or previous issues that clarifies what the intended behaviour of the two situations below is, except for #6 which doesn't really help. The diagnostic function doesn't say anything is amiss about slacks in either case.

What is is the intended behaviour for these two situations?

A fully connected network with multiple external grids seems to result in one of them being ignored:
``` python
>>> import pandapower as pp
>>> import pandapower.networks as ppnw
>>> net = ppnw.case9()
>>> pp.create_ext_grid(net, 3)
1
>>> pp.runpp(net)
>>> print(net.res_ext_grid)
        p_mw     q_mvar
0   0.000000   0.000000
1  71.912983  27.784276
```

Multiple islanded sub-networks, where each sub-network has an external grid, results in an error:
``` python
>>> import pandapower as pp
>>> import pandapower.networks as ppnw
>>> net = ppnw.case9()
>>> b = pp.create_bus(net, vn_kv=100)
>>> pp.create_ext_grid(net, b)
1
>>> pp.create_load(net, b, p_mw=10)
3
>>> pp.runpp(net)
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/run.py", line 225, in runpp
    _powerflow(net, **kwargs)
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/powerflow.py", line 74, in _powerflow
    result = _run_pf_algorithm(ppci, net["_options"], **kwargs)
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/powerflow.py", line 151, in _run_pf_algorithm
    result = _run_newton_raphson_pf(ppci, options)
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/pf/run_newton_raphson_pf.py", line 51, in _run_newton_raphson_pf
    ppci, success, iterations = _run_ac_pf_without_qlims_enforced(ppci, options)
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/pf/run_newton_raphson_pf.py", line 131, in _run_ac_pf_without_qlims_enforced
    V, success, iterations, J, Vm_it, Va_it = newton(Ybus, Sbus, V0, pv, pq, ppci, options)
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/pypower/newtonpf.py", line 87, in newtonpf
    F = _evaluate_Fx(Ybus, V, Sbus, pv, pq)
  File "/Users/michael/Dropbox/mac_git/pandapower/pandapower/pypower/newtonpf.py", line 132, in _evaluate_Fx
    mis = V * conj(Ybus * V) - Sbus
  File "/Users/michael/miniconda3/envs/pandapower/lib/python3.7/site-packages/scipy/sparse/base.py", line 499, in __mul__
    raise ValueError('dimension mismatch')
ValueError: dimension mismatch
```

lthurner, 2020-05-25:
> A fully connected network with multiple external grids seems to result in one of them being ignored:

No they are both active. There are just no loads between your two ext_grids, so that the power is supplied completely by one ext_grid. The current takes the path of least resistance, so it always flows through the first ext_grid. If you connect the ext_grid at bus 8 (the opposite side of the network), you will see that bot ext_grids are active:

>>> import pandapower as pp
>>> import pandapower.networks as ppnw
>>> net = ppnw.case9()
>>> pp.create_ext_grid(net, 8)
1
>>> pp.runpp(net)
>>> print(net.res_ext_grid)
        p_mw     q_mvar
0   14.602316  -1.980213
1  56.953635  40.032986

> Multiple islanded sub-networks, where each sub-network has an external grid, results in an error:

This shouldn't happen, it is a bug. Might have to do something with the single bus island network. If you connect a line to the second island, i suspect it will converge (haven't tried it yet though)

jurasofish, 2020-05-25:
You're right! Adding a line sorted it out, thanks for the help

```python
>>> import pandapower as pp
>>> import pandapower.networks as ppnw
>>> net = ppnw.case9()
>>> b1 = pp.create_bus(net, vn_kv=100)
>>> b2 = pp.create_bus(net, vn_kv=100)
>>> pp.create_line_from_parameters(net, b1, b2, 1, 1, 1, 0, 1)
9
>>> pp.create_load(net, b2, p_mw=10)
3
>>> pp.create_ext_grid(net, b1)
1
>>> pp.runpp(net)
>>> print(net.res_ext_grid)
        p_mw     q_mvar
0  71.954702  24.068958
1  10.010020   0.010020
```

lthurner, 2020-05-25:
It should still work even with a single bus island, so leets keep this open. Maybe while you are on it you can add it as a failing test so we don't forget about the issue?

jurasofish, 2020-05-25:
Sure thing, good idea

jurasofish, 2020-06-18:
as a note this works when numba isn't installed

SimonRubenDrauz, 2020-11-04:
I had a look a it and could find out that, as jurasofish already mentioned, in case numba is installed it is not working. The reason, why it is working then, is, that instead of numba, pypower is used.

When comparing both methods I found out that in Ybus matrix looks differently. In case of pypower, one has a 10x10 matrix, in case of numba it is a 10x9 matrix. This causes the error, as the matrix is not symmetric.

The bid difference between both is that in case of pypower the shape is passed:
[image]
while in numba it is not:
[image]

I tried to find out how it determines the shape then. Actually, the entries given are at first the values, then the column and then finally where a new row starts. Thus, by the maximum column number and the number of new starting points, it determines the shape. As the maximum column number is 8 in this case, the shape is 10x9 and not the expected 10x10. (Thus column 10 is empty --> I checked it, it also is the case for pypower)

This could be solved by just passing the shape:
[image]

I compared the results in pypower and numba and it looks good. I also ran all tests and they passed as well. 

However, as I didn't have the time to get too deep into the mathematics, I just want to ask you @lthurner, if the implementation is correct. You find the corresponding PR here: #976

closed 2020-12-03 by rbolgaryn