e2nIEE/pandapower #781: Simple Plot

bmitrauncc, 2020-05-25:
If I create a network using custom data, I am unable to get the plot using pp.simple_plot(net), it returns
"The library igraph is selected for plotting, but not installed correctly".

But if I load any pre-saved network and use the command I can get the network plotted.

jurasofish, 2020-05-25:
``pip install python-igraph`` to install igraph, then it should work.

The reason the pre-saved networks work is probably because they have pre-existing geometry - check if ``net.bus_geodata`` and/or ``net.line_geodata`` contain geometry. If there's no pre-existing geometry then pandapower uses igraph (by default) to generate artificial geometry.

Perhaps it would be worth including matplotlib and python-igraph by default. I had the same issue recently and pip installed igraph, which was the wrong thing to do and that took me half an hour to figure out. Including by default would really improve user experience imo, but at a slightly increased risk of installation issues.

lthurner, 2020-05-26:
The reason we don't include this in the dependencies is because 

> pip install python-igraph

doesn't work on windows. It used to be only possible to install this package on windows through the prebuilt wheels from https://www.lfd.uci.edu/~gohlke/pythonlibs/. Now it seems that at least it is possible to install python-igraph through conda, so

> conda install python-igraph

should work. Still, if we include python-igraph into the dependencies, pip install will fali on windows unless python-igraph was installed through conda first. I am not sure what the best approach is here, both solutions aren't perfect...

jurasofish, 2020-05-26:
I suspected it might be an installation issue.

>  https://www.lfd.uci.edu/~gohlke/pythonlibs/.

love this site, an absolute blessing to the community

I've just put through a pull request to make things a bit clearer, let me know what you think

lthurner, 2020-05-26:
Looks good, merged, thanks.

@bmitrauncc I think the issue was explained, so I hop this answers your question. Please reopen if you still have questions.

closed 2020-05-26 by lthurner

bmitrauncc, 2020-05-27:
I did uninstall python-igraph and reinstall using conda, but the issue still persists.