e2nIEE/pandapower #784: Jacobian

HM052020, 2020-05-26:
Hi everybody,

How can I get the Jacobian Matrix when running PF with methods other than NR (e.g. fdxb/fdbx/...) in pandapower?
By "net._ppc["internal"]["J"].todense()", I can not extract the Jacobian Matrix when I set "fdxb" or "fdbx" method in "pp.runpp(net,'fdxb')" command with "pp" as pandapower.
I'm fresh in working in pandapower and also not an expert in the electric power system.
So, I will be happy if the pandapower developers or any other one could guide me.

Best regards,
HM

lthurner, 2020-05-28:
The fast decoupled power flow doesn't use the jacobian matrix, it uses other matrices. Thats why you will only find the Jacobian after a Newton Raphson.

DRL2019, 2020-05-28:
Dear Leon,

Thank you for your reply.
So, is there any possibility in pandapower to extract that matrix, as you mentioned in above and I call it the charatristic matrix, for the fd method?
Thank you in advance for your reply.

Best regards,
Hodjat

DRL2019, 2020-05-28:
Dear Leon,

Thank you for your reply.
So, is there any possibility in pandapower to extract that matrix, as you mentioned in above and I call it the charatristic matrix, for the fd method?
Thank you in advance for your reply.

Best regards,
HM

marhofmann, 2020-06-03:
Check out this tutorial:
https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

DRL2019, 2020-06-05:
> Check out this tutorial:
> https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

Dear Martin @marhofmann 
Thank you for your reply.
I looked at this link but it is about the Newton PowerFlow Solver. My question is about other solver like as Fast-decoupled. And I wanted to extract the charactristic matrix for this solver by pandapower. I'm not expert in pandapower and also in electric power pystem ...

Best,

SteffenMeinecke, 2021-11-10:
This issue seems not significantly relevant to many users so the internal matrices other than with the newton-raphson algorithm are not available from the outside of pandapower. If you want to have a look into it, you can debug to the function`_run_pf_algorithm()` in `powerflow.py`, where the different algorithms are called, and its sub functions . In the case of Fast-decoupled load flow you can have a look to `_call_power_flow_function()` in `runpf_pypower.py` where alg == 2 or alg == 3.

closed 2021-11-10 by SteffenMeinecke