e2nIEE/pandapower #787: Issue adding Load(s) to Cigre_LV_network

12o13, 2020-05-27:
After setting up the Cigre_LV_network I attempt to add a load and get an error which results in the load being partially added. I am trying to add a unique Load(s) to the network. 

```
LVGrid = pn.create_cigre_network_lv()
pp.runpp(LVGrid)
pp.create_load(LVGrid, bus=5, p_mw=BSS, q_mvar=0, scaling=1, name="HouseLoad")
```

The output I get is 

```
LVGrid.load
Out[36]: 
         name   bus     p_mw    q_mvar  ...  sn_mva  scaling  in_service  type
0     Load R1   2.0  0.19000  0.062450  ...     NaN      1.0        True  None
...
13   Load C19  42.0  0.01440  0.006974  ...     NaN      1.0        True  None
14   Load C20  43.0  0.00720  0.003487  ...     NaN      1.0        True  None
15  HouseLoad   5.0      NaN       NaN  ...     NaN      NaN         NaN   NaN
```

So you can see the Load was added on index Bus 5 which is technically what I want to do, but none of the Load info was attached after the bus index failed. 

What is the proper way to attach a new load, preferably by name? And how do I avoid this error moving forward?

lthurner, 2020-05-27:
I can't reproduce this issue, for me it adds the load correctly. Which version are you working with? What kind of an error are you getting?

12o13, 2020-05-27:
I think I figured out what was causing the issue: I was trying to pass a series to my p_mw value. That would seem to explain the error flag:
```
...
  File "C:\Users\NAME\anaconda3\lib\site-packages\pandas\core\internals\managers.py", line 442, in apply
    applied = getattr(b, f)(**kwargs)
  File "C:\Users\NAME\anaconda3\lib\site-packages\pandas\core\internals\blocks.py", line 914, in setitem
    values[indexer] = value
ValueError: setting an array element with a sequence.
```
I imagine I'll be more aware of what is causing this issue as I start digging into the time-series stuff more. I'll leave this open for another response incase passing a series is something that is supposed to work already. After you reply and then I'll close if necessary.

closed 2020-05-27 by 12o13